stm32-rs/stm32-rs #9: Unable to modify svd cpu node

tones111, 2018-04-27:
Looking through svdpatch.py it doesn't appear that it's able to handle modifcations to cpu node children.

For example, stm32f303.svd contains
<cpu>... <nvicPrioBits>3</nvicPrioBits>...</cpu>

However, the reference manual (RM0316, Section 14.1.1) states "4 bits of interrupt priority are used"

adamgreig, 2018-04-28:
You're right, at the moment it doesn't have any way to modify things outside of the peripheral blocks, but clearly there are bugs everywhere that will need patching! I'll think about how to address this.

tones111, 2018-04-29:
It would seem to me that there are several groups that would be interested in maintaining changes to svd files.  Especially if svds are being used to generate for other architectues.  There's no reason we should all be doing the same task differently.

@japaric
It might make more sense to discuss tooling at the broader embedded rust level.  If we could agree on the config file format then only one tool would need to be developed for the entire ecosystem.

The downside of the approach here is that it requies external python dependencies which doesn't play well with cargo builds.  Moving the implementation into a rust library would work really well for build.rs usage.

therealprof, 2018-06-02:
@adamgreig Any news about this? I'd need to add missing interrupts to `stm32f7x7`...

adamgreig, 2018-06-03:
@therealprof that's a similar but distinct problem around editing/adding `<interrupt>` nodes inside a `<peripheral>`, right?

therealprof, 2018-06-03:
@adamgreig Yes, that's right.

adamgreig, 2018-06-05:
I've roughed out an extension to the DSL to permit editing the cpu node and interrupts etc, shown [here](https://github.com/adamgreig/stm32-rs/compare/extend-dsl). It's back-compat with all the existing stuff but should make it easy to change anything else in the SVD. I'll hopefully get it implemented tomorrow evening; any thoughts?

adamgreig, 2018-06-10:
@therealprof could you open another issue (or PR) for the missing interrupts, or point me to which ones they are? It should now be possible to add them.

closed 2018-06-10 by adamgreig

therealprof, 2018-06-17:
Sure, it's a bit tricky to figure out which ones are missing but I've already identified a few.

commit de117ef6 (2018-06-10): Fix nvicPrioBits on many STM32s, closes #9