stm32-rs/stm32-rs #10: Update peripheral comparison tables

bergus, 2018-04-29:
Hi, I am looking at the [peripheral register support comparison tables](https://agg.io/u/stm32/stm32l/index.html) you mentioned [here](https://github.com/japaric/embedded-hal/issues/61#issuecomment-372839347). Unfortunately they don't include the STM32L4 devices, though they are contained in this repository. I'm not sure whether this is the right place to report the issue, but could you update the pages?

adamgreig, 2018-04-29:
Are [these](https://agg.io/u/stm32/stm32l4/index.html) what you're looking for?

bergus, 2018-04-29:
I do wonder though how you generated the addresses in these tables? They often collide between multiple peripherals of the same device.

adamgreig, 2018-04-29:
They're made using [htmlcomparesvd.py](https://github.com/adamgreig/stm32-rs/blob/master/scripts/htmlcomparesvd.py) but you're right, it looks like it incorrectly copies the base address for peripherals that are derived from one another.

adamgreig, 2018-04-29:
Fixed in #04f928e and new versions uploaded. Thanks for spotting!