stm32-rs/stm32-rs #19: Use form to split device files into mutiple folder.

gbip, 2018-05-24:
I think it is a good idea to use form to split devices files (`stm32f429.rs`) into a folder with subfolder for each module.
I can implement it if you want to :smile:

gbip, 2018-06-03:
Hey @adamgreig can you take a look at this :smile:
I am willing to implement it if you think that it can make documentation generation easier !

adamgreig, 2018-06-03:
If you'd like to have a go I'm happy to review, I'm up for at least seeing what it looks like and happy to merge if it seems reasonable. If it makes documentation easier then even better!

I imagine we'd have a folder for each device, so like `stm32f4/src/stm32f405/` which contains the generated files, and then feature-gated modules in `lib.rs` as currently?

adamgreig, 2018-06-11:
The support for svd2rust 0.13.1 means each device already gets its own folder, so hopefully running Form over it should be even easier now. I'll have a play with it this week.

gbip, 2018-06-11:
I'll try to see how it plays out this week too.

gbip, 2018-06-15:
So since I am not familiar at all with Makefiles, I didn't do anything except running form on a few `mod.rs` with success (by hand).

adamgreig, 2018-07-11:
Closed by #29

closed 2018-07-11 by adamgreig