stm32-rs/stm32-rs #21: Update to latest svd2rust?

phil-opp, 2018-06-01:
The current svd2rust version is 0.13.1. Could you update the crates.io version to the newest svd2rust version to make it compatile with [cortex-m-quickstart](https://docs.rs/cortex-m-quickstart/0.3.1/cortex_m_quickstart/)?

I tried to regenerate the crates myself, but I got a lot of `duplicate definitions with name x` errors.

therealprof, 2018-06-02:
svd2rust 0.13.1 works quite differently from previous versions. The output is not a single `.rs` file to stdout anymore but 3 files directly written to the current directory. While the changes to pick up the new main register description are straight forward this will need a bit of Makefile magic to not break parallel builds and then what to do with the new extra files?

phil-opp, 2018-06-02:
As far as I understand it the additional two files are a `build.rs` build script and a linker script which ensures that the default exception handler is called for unregistered interrupts.

therealprof, 2018-06-02:
Yes, and somehow they will need to be integrated...

adamgreig, 2018-06-03:
Yea, I've been playing with the new svd2rust but haven't yet figured out a good way to make it work, especially with multiple svd2rust-generated devices per family crate. Open to suggestions...

therealprof, 2018-06-03:
@adamgreig Well, the `build.rs` is really just static. It checks whether the `rt` feature flag is set and if so it'll spit out the `device.x` file in the compilation path. In theory it should suffice to have special version of the `build.rs` that looks at the target device and the `rt` feature flag, and a set of device specific `device.x` files (we can simply rename them while placing and place them next to peripheral register code). The `build.rs` could then pick up the device specfic `.x` file and rename it back to`device.x` in the  output directory.

adamgreig, 2018-06-10:
I've pretty much got this all working in #24 but need to test it with some actual applications and ran out of time. Feedback welcome if anyone wants to take a look. There were quite a few changes in how applications are built with the latest cortex-m, cortex-m-rt, and svd2rust, so most of my example applications are now broken too...

adamgreig, 2018-06-11:
Whew, okay, after finding the new `device` feature of the `cortex-m-rt` crate it all works and I can build some test applications. I'm merging this in and will release updated crates this week once I've had a chance to test them a bit more.

closed 2018-06-11 by adamgreig

therealprof, 2018-06-11:
@adamgreig Nice, thanks! I'll check that out in a bit...