stm32-rs/stm32-rs #22: changes to Makefile

baldwin1974, 2018-06-10:
Hi there, I use a mac which has a cantankerous version of sed. Also, `svd2rust` has changed a bit somehow, or at least it is different on mac. `svd2rust` seems to not need a pipe anymore, but it only creates a lib.rs file now. Thus, I think your Makefile could be changed a bit without causing harm to users of other OSs:

(1) add a `'.tmp'` after `-i` flag in the sed command
`sed -i.tmp '1,3d;5,11d;13,16d;23s/interrupt/self::interrupt/'
`
(2) modify the svd2rust line to be three lines:
`svd2rust -i $<
`mv lib.rs $@
`rustfmt $@

Excerpt from a page (http://www.grymoire.com/Unix/Sed.html#uh-62h) on sed: 
"The GNU version of sed allows you to use "-i" without an argument. The FreeBSD/Mac OS X does not. You must provide an extension for the FreeBSD/Mac OS X version."

adamgreig, 2018-06-10:
Good timing, I'm in the midst of updating the Makefile to support the latest svd2rust (#21). It's a bit more complicated than your suggestion because of supporting parallel builds, and also a custom `build.rs` needs to be written and the top level `lib.rs` updated, but I'm getting there.

The OS X sed thing is annoying. It looks like `-i ''` should do the same as the GNU version?

baldwin1974, 2018-06-10:
Yes, it is silly handholding by Apple. -i'' does the same thing, yes.

closed 2018-06-10 by baldwin1974