stm32-rs/stm32-rs #23: Trying to manage a "derivedFrom" peripheral

baldwin1974, 2018-06-10:
I hate to pester you on late on a Sunday night, but I wonder if you can help me a bit more.  I am seeing the same errors I referred to in the svd2rust issue that I closed. For example, it seems to me it should be possible to do this:

```
let p = stm32h7::stm32h7x3::Peripherals::take().unwrap()
let gpiob = p.GPIOB;
gpiob.gpiob_moder.write ( |w| w.mode7.bits(0) );
```

However, this generates a predictable error: there is no gpiob_moder field within GPIOB (because it is derivedFrom GPIOA, which *does* have a gpioa_moder field, naturally). What am I missing?

adamgreig, 2018-06-10:
No worries, I'm working on stm32-rs tonight anyway.

You're hitting what is actually an error/bug in the SVD file: GPIOA's registers should be named `MODER` not `GPIOA_MODER`; as such since `GPIOB` is derived from `GPIOA`, it too has `GPIOA_MODER`.

For right now, you could just do:
```rust
gpiob.gpioa_moder.write(|ẁ| w.mode7.bits(0));
```
but of course this is very unsatisfactory. The solution is to fix the SVD file. You could make a PR to do this, or I can have a go later. It would look a lot like the one for the F7 which has a similar issue: https://github.com/adamgreig/stm32-rs/blob/master/devices/stm32f7x7.yaml#L152-L156

Basically you'd tell it to patch GPIOA so that GPIOA_MODER becomes MODER, and etc for all afflicted registers.

baldwin1974, 2018-06-10:
It is worth noting also that this *does* compile:
```
gpiob.gpioa_moder.write ( |w| unsafe { w.mode7.bits(0) } );
```

(Note it is now gpioa_moder.) But I do not think it works in practice, or even should work. Why would it, since gpiob is a different register block than gpioa? 

As an aside, I have noticed that other crates (for example stm32f4/stm32f407) have gpioa and gpiob modules (and in fact the outlier gpioi module, though no gpioc, gpiod, gpioe, etc.). And not surprisingly the gpiob module has a moder field. I seems likely that STMicro has just not executed their svd files uniformly, and svd2rust can only do so much.

baldwin1974, 2018-06-10:
Ah, our comments crossed paths. Understood, I will take a shot at revising the svd file.

baldwin1974, 2018-06-10:
By the way, what do you mean when you say "tell it" to patch GPIOA. What am I telling that to?

I am obviously a newb at git, so that may be my problem?

closed 2018-06-10 by baldwin1974

adamgreig, 2018-06-10:
Ah, sorry, I should have been clearer. This whole repository basically organises patching SVD files; there's a preprocessor which takes the SVD from the vendor and a set of patches (in YAML format) and applies them to make a new SVD, which is then fed in to svd2rust. You'd be updating the "device YAML" file for the H7x3 to tell the patching script to perform additional modifications on the vendor SVD.

baldwin1974, 2018-06-10:
Man, that is cool, I didn't grasp that before. Ok, I will look through the device files to try and get those changes worked out.

baldwin1974, 2018-06-10:
I got no clue how to submit a pull request. I'll learn eventually. Here is the code:

```
"GPIOA":
  _modify:
    # the SVD misnames the GPIO fields, as can be seen below. Fix it.
    GPIOA_MODER:
        name: MODER
    GPIOA_OTYPER:
        name: OTYPER
    GPIOA_OSPEEDR:
        name: OSPEEDR
    GPIOA_PUPDR:
        name: PUPDR
    GPIOA_IDR:
        name: IDR
    GPIOA_ODR:
        name: ODR
    GPIOA_BSRR:
        name: BSRR
    GPIOA_LCKR:
        name: LCKR
    GPIOA_AFRL:
        name: AFRL
    GPIOA_AFRH:
        name: AFRH
```

This succeeded. I will be going through the datasheet over the next few weeks so I can try to police the svd further as I go.

bergus, 2018-06-11:
@baldwin1974 https://help.github.com/articles/about-pull-requests/ :-)

adamgreig, 2018-06-12:
I've included your patch above, thanks.