stm32-rs/stm32-rs #25: new sed script issues

baldwin1974, 2018-06-11:
Hi there again, I think for mac os the sed command `sed -i ""` is the necessary format. In your newest Makefile from last night, the command `sed -i''` (in single quotes) generates errors. Sorry about giving bad information earlier. I do think using `-i.tmp` is better but it does generate more bloat, and using double-quotes works; there must be a space between -i and the double-quotes, too.

adamgreig, 2018-06-11:
I don't think `-i ''` can work, at least with GNU sed, as it interprets the '' as an empty command string and the actual command string as an invalid filename. Perhaps `--in-place=""` will work on both?

baldwin1974, 2018-06-11:
I just tried `--in-place=""` on my mac and it didn't work. Are you saying neither double-quotes nor single quotes will work if there is a space between them and the `-i`? If so, then the only solution I can think of for both linux and mac is to run `-i.tmp` and then delete the .tmp file, or you could just put a bit of guidance in the Makefile ...

adamgreig, 2018-06-12:
I've updated the Makefile to not use `-i` at all; could you check if this works for you please?