stm32-rs/stm32-rs #27: Missing interrupts for STM32F7x7

therealprof, 2018-06-17:
As mentioned in #9 a couple of interrupts are missing for this MCU:
- ETH (=61)
- ETH_WKUP (=62)
- PVD (=1)
- TIM1_UP_TIM10 (=25)
- TIM1_CC (=27)
- I2C2_EV (=33)
- I2C2_ER (=34)
- SDMMC1 (=49)
- UART4 (=52)
- USART6 (=71)
- I2C3_EV (=72)
- I2C3_ER (=73)
- CRYP (=79)
- UART7 (=82)
- LCD-TFT (88 or 89, don't know which one)
- SAI2 (=91)
- QuadSPI (=92)
- LP Timer1 (=93)
- HDMI-CEC (=94)
- I2C4_EV (=95)
- I2C4_ER (=96)
- SPDIFRX (=97)
- DFSDM1_FLT0 (=99)
- DFSDM1_FLT1 (=100)
- DFSDM1_FLT2 (=101)
- DFSDM1_FLT3 (=102)

adamgreig, 2018-06-18:
Thanks for putting together this list. It looks to me like many (all?) are also missing on STM32F7x5, 6, and 9, as well (but not x2 or x3..). I'll double check and then add them all.

adamgreig, 2018-06-21:
Now done. See [f7_interrupts.yaml](https://github.com/adamgreig/stm32-rs/blob/master/devices/common_patches/f7_interrupts.yaml) and [f7_interrupts_extra.yaml](https://github.com/adamgreig/stm32-rs/blob/master/devices/common_patches/f7_interrupts_extra.yaml).

commit a2e9fff9 (2018-06-21): Add missing interrupts to STM32F7 devices. Fixes #27
closed 2018-06-21 by adamgreig