stm32-rs/stm32-rs #35: Missing PLLSAI fields in RCC_CR register for stm32f7x6

phil-opp, 2018-07-11:
The reference manual RM0385 for STM32F75xxx and STM32F74xxx mentions `PLLSAION` and `PLLSAIRDY` bits at bit 28 and 29 of the RCC clock control register (RCC_CR). We need those bits to initialize the LCD controller of our stm32f746 discovery board. However, they are missing from both the SVD file and the generated code.

Previously we used the SVD of the AdaCore project, which can be found here: <https://github.com/AdaCore/svd2ada/blob/master/CMSIS-SVD/ST/STM32F7x.svd>. This file contains the PLLSAI fields, so I maybe they applied custom patches? (I tried to look through the history of that file, but couldn't find any addition of the PLLSAI bits.)

adamgreig, 2018-07-11:
Looks like the `7x2` and `7x3` SVDs have these (they are slightly newer from ST and generally have more fixes) and the `7x5`, `7x6`, `7x7`, `7x9` do not.  I'll add them now.. are you using released crates or a local path (i.e. do you need a new release with this soon?).

phil-opp, 2018-07-11:
Thanks a lot!

I'm using the released crate. For now I worked around it by manipulating the bits directly through the `bits` method, so there is no urgency. Could you just ping me when you release a new version so that I don't forget to remove this workaround? :D

closed 2018-07-11 by adamgreig

adamgreig, 2018-07-11:
@phil-opp I've published 0.2.2 since I wanted to try out having docs.rs build for more of the crates now, but it includes the PLLSAION/PLLSAIRDY fields too.

phil-opp, 2018-07-12:
Thanks a lot, works perfectly!

commit 7cd31904 (2018-07-11): Add PLLSAIRDY PLLSAION to STM32F7x{5,7,8,9}. Closes #35.