stm32-rs/stm32-rs #36: STM32F4: USARTxEN, but UARTxRST

SillyFreak, 2018-07-13:
I have encountered this in several STM32F4 svd files (not looked at other series): several fields would be named `UARTxRST`, but their `EN` counterpart as well as the field's description would say `USART`, note the `S`.

There are several legitimate `UARTxRST` fields (with matching description and `EN`), and which `x`s are wrong varies by device. Also I've never written an svd patch, that's why this is not a pull request, but I'd be happy to provide one with a little guidance! What structure would you suggest for the patch? one per `x` to be included in the device files, or something else?

adamgreig, 2018-07-15:
Thanks for reporting this, and good catch. Annoyingly it looks like the reference manual makes the same errors (e.g. in RM0090 for STM32F405/07/15/17, APB1RSTR has UART2RST and UART3RST instead of USART). So far the general policy has been to fix the SVD to match the reference manual. In this case I think the reference manual is pretty clearly wrong too but it's the best source we have for the official name of those fields...

If one was going to fix this, probably you'd make a patch in `devices/common_patches/f4_rcc_usart.yaml` with something like...

```yaml
RCC:
  APB1RSTR:
    _modify:
      UART2RST:
        name: USART2RST
      UART3RST:
        name: USART3RST
```

(and any others as required), and then include it on the relevant device files.

Can you find any examples where the SVD is wrong like this but the reference manual is correct?

SillyFreak, 2018-07-18:
OK, I have started investigating this.

```
cd svd/
./extract.sh
# look for mismatches between name and description, which appear in consecutive lines
pcre2grep -M '(<name>UART.*\n.*USART)|(<name>USART.*\n.*UART)' *.svd | tr -d ' \r\n' | sed 's/stm/\nstm/g' > errors
```

this gives me 83 matches in the f2, f4, f7, l0, l1 series. most are `UARTxRST`, but there are others such as `UARTxSEL`, `USARTx[SM]EN` (UARTx clock enable [during sleep mode] bit), and a lone `USART4` (UART4 global interrupt).

Beginning to compare the f2 series with RM0033 p.108, I see this:

```
 stm32f215.svd:<name>UART5RST</name><description>USART5reset</description>
 stm32f215.svd:<name>UART4RST</name><description>USART4reset</description>
!stm32f215.svd:<name>UART3RST</name><description>USART3reset</description>
!stm32f215.svd:<name>UART2RST</name><description>USART2reset</description>
 stm32f217.svd:<name>UART5RST</name><description>USART5reset</description>
 stm32f217.svd:<name>UART4RST</name><description>USART4reset</description>
!stm32f217.svd:<name>UART3RST</name><description>USART3reset</description>
!stm32f217.svd:<name>UART2RST</name><description>USART2reset</description>
```

Lines with a ` `: consistent with RM; lines with a `!`: RM has it correct. So yes, this does definitely exist. Others remain to be checked for now.