stm32-rs/stm32-rs #40: When not specifying the "rt" flag, an extra 600 bytes of vector table is added

korken89, 2018-07-30:
Hi, I seem to have found a bug that when not having the `rt` flag in features, 600 bytes are added in the vector table where each vector is pointing to the `DefaultHandler`.

I have tested with the following 2 `Cargo.toml` configurations:
```
[dependencies]
panic-abort = "*"
cortex-m-rt = "*"
cortex-m = "*"

[dependencies.stm32f4]
version = "0.2.0"
features = ["stm32f405", "rt"]
```
and

```
[dependencies]
panic-abort = "*"
cortex-m-rt = "*"
cortex-m = "*"

[dependencies.stm32f4]
version = "0.2.0"
features = ["stm32f405"]
```
where the later is consistently 600 byte larger.

This was tested on the latest nightly. 
Please tell me if there is something more you want me to test or if there is some further information that you need.

adamgreig, 2018-07-30:
Thanks for the bug report! This is actually expected behaviour of the `cortex-m-rt` crate: if you don't enable its `device` feature, it fills all possible vector table entries with DefaultHandler, to ensure the resulting binary can run on any theoretical Cortex-M. When you enable `device`, it only emits the required size of vector table, which is smaller for the STM32 (and all other devices I know of).

See [cortex-m-rt docs](https://docs.rs/cortex-m-rt/0.5.1/cortex_m_rt/#device) for details.

adamgreig, 2018-07-30:
(and in case it wasn't clear: enabling `rt` on the `stm32f4` crate causes `cortex-m-rt`'s `device` feature to be enabled, hence the observed behaviour).

closed 2018-07-30 by adamgreig

korken89, 2018-07-31:
Thank you for explaining and the link to the docs! The issue arose for me when having to use my own `cortex-m-rt` crate to add `pre_init` functionality, which of course collided with the `cortex-m-rt` of this lib.

For now I will live with the huge vector table, until the patch is merged in `cortex-m-rt`.