stm32-rs/stm32-rs #47: How to merge registers fields and modify its content?

octronics, 2018-09-13:
I want to merge the CNT_H and CNT_L 16bit field into one 32bit CNT field. I found the `_merge` but it reduce the name as `CNT_`. How can I remove the trailing underscore?
I found the `_modify` and try to update the description of the field, to reflect its meaning and not was as been merged (the CNT_H description). But this doesn't work. The README says that _modify is done first.
```
"TIM2":
  CNT:
    _merge:
      - "CNT*"
    # Actualy a 32bit register
    _modify:
      "CNT_":
        name: CNT
        description: Counter value
        value: [0, 4294967295]
```
Would you please help me?

adamgreig, 2018-09-16:
This is a bit tricky. Maybe you could use `_modify` first to name them `CNTH` and `CNTL`, and then use `_merge` to combine them without the trailing underscore?

octronics, 2018-09-17:
Thanks. Will give it a try.

octronics, 2018-09-17:
I come to that and it works. Feel free to close the issue.
```
"TIM2":
  CNT:
    _modify:
      CNT_H:
        name: CNTH
        description: Counter value
      CNT_L:
        name: CNTL
    _merge:
      - "CNT*"
    CNT: [0, 4294967295]
```

adamgreig, 2018-09-17:
Glad it worked. Not ideal but it will do.

closed 2018-09-17 by adamgreig