stm32-rs/stm32-rs #56: Cluster and register array support

burrbull, 2018-09-17:
I have 2 questions important for me:
- Do you plan to add support such SVD items as clusters and register arrays.
- If answer is `yes` or `maybe`. What if I propose to use other, already patched files instead of official as source ([stm32svd-compare](https://github.com/burrbull/stm32svd-compare)).
What I change? My idea is simple. If you enter `All` folder, you'll see SVD with cutout peripheral contents.
I compare if they are identical (I use hash for speed up, naming and easy restore). Then I compare differences manually with difftool (`meld` is better). Most often these are only description differences. But when I fix them, I can see real issues. So I fixed a lot of typos, unify some field/register names and spent a lot of time. But the biggest change is grouping some registers to clusters (like in CAN, DMA, backup registers an other).

adamgreig, 2018-09-17:
I would like to have cluster and array support but have no current plans to implement it myself.

I'm not sure about using already patched files as the sources of truth though. I would rather always start from the official ST files and then apply patches. Maybe it would be possible to create YAML descriptions of the changes from your patched files, and then use those to improve the existing patches?

burrbull, 2018-09-18:
I can try to implement this. But I need to know your vision of API: how to describe clusters, how to move, remove registers and maybe something else.

adamgreig, 2018-12-16:
Closed by #112.

closed 2018-12-16 by adamgreig