stm32-rs/stm32-rs #63: Failed to build with 1.30.0-nightly

sprhawk, 2018-09-27:
environment:
rustc: 1.30.0-nightly (4591a245c 2018-09-22)
stm32f3 version: 0.2.3
features = ["stm32f302", "rt"]

```
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

A lib I use svd2rust to generate from [STM32F302.svd](https://github.com/posborne/cmsis-svd/tree/master/data) can be built

adamgreig, 2018-09-30:
Could you post your full `cargo.toml` and perhaps a minimal `src/main.rs` that replicates the issue? I've just tried a minimal new project which builds fine. Maybe you don't have a `build.rs` script in your project at all? Here's the very very minimal project that builds fine:

cargo.toml:
```toml
[package]
name = "stm32test"
version = "0.1.0"
authors = ["Adam Greig <adam@adamgreig.com>"]

[dependencies]
cortex-m = "0.5.7"
cortex-m-rt = "0.6.4"
panic-abort = "0.3.0"

[dependencies.stm32f3]
version = "0.2.3"
features = ["stm32f303", "rt"]
```

build.rs:
```rust
use std::env;
use std::fs::File;
use std::io::Write;
use std::path::PathBuf;

fn main() {
    let out = &PathBuf::from(env::var_os("OUT_DIR").unwrap());
    File::create(out.join("memory.x"))
        .unwrap()
        .write_all(include_bytes!("memory.x"))
        .unwrap();
    println!("cargo:rustc-link-search={}", out.display());
}
```

.cargo/config:
```toml
[target.thumbv7em-none-eabihf]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
  "-C", "linker=arm-none-eabi-ld",
  "-Z", "linker-flavor=ld",
  "-Z", "thinlto=no",
]

[build]
target = "thumbv7em-none-eabihf"
```

memory.x:
```
MEMORY
{
    FLASH : ORIGIN = 0x08000000, LENGTH = 1M
    RAM :   ORIGIN = 0x20000000, LENGTH = 128K
}
```

src/main.rs:
```rust
#![no_std]
#![no_main]
extern crate cortex_m;
extern crate cortex_m_rt;
extern crate panic_abort;
use cortex_m_rt::entry;
extern crate stm32f3;
use stm32f3::stm32f303;

#[entry]
fn start() -> ! {
    let p = stm32f303::Peripherals::take().unwrap();
    let gpioa = &p.GPIOA;
    gpioa.odr.modify(|_, w| w.odr0().set_bit());
    loop {
        cortex_m::asm::nop();
    }
}
```

sprhawk, 2018-10-01:
Curiously, your code built, but mine not. 

Just replace stm32f3 with stm32f302 I generated, it built.
maybe I'm missing something?

Cargo.toml

```
[package]
authors = ["sprhawk <465558+sprhawk@users.noreply.github.com>"]
edition = "2018"
name = "esc"
version = "0.1.0"

[dependencies]
cortex-m-semihosting = "0.3.1"
panic-halt = "0.2.0"
# stm32f302 = { path = "../stm32f302/" }

# Uncomment for the panic example.
# panic-itm = "0.4.0"

# Uncomment for the allocator example.
# alloc-cortex-m = "0.3.5"

# Uncomment for the device example.
# [dependencies.stm32f30x]
# features = ["rt"]
# version = "0.7.1"

[dependencies.cortex-m]
version = "0.5.7"
features = ["inline-asm"]

[dependencies.cortex-m-rt]
version = "0.6.3"

[dependencies.panic-semihosting]
version = "0.5.0"

[dependencies.stm32f3]
version = "0.2.3"
features = ["stm32f302", "rt"]

# this lets you use `cargo fix`!
[[bin]]
name = "esc"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
```

.cargo/config

```
# [target.thumbv7m-none-eabi]
[target.thumbv7em-none-eabihf]
# uncomment this to make `cargo run` execute programs on QEMU
# runner = "qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel"

[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# uncomment ONE of these three option to make `cargo run` start a GDB session
# which option to pick depends on your system
runner = "arm-none-eabi-gdb -q -x openocd.gdb"
# runner = "gdb-multiarch -q -x openocd.gdb"
# runner = "gdb -q -x openocd.gdb"

rustflags = [
  # LLD (shipped with the Rust toolchain) is used as the default linker
  "-C", "link-arg=-Tlink.x",

  # if you run into problems with LLD switch to the GNU linker by commenting out
  # this line
  # "-C", "linker=arm-none-eabi-ld",

  # if you need to link to pre-compiled C libraries provided by a C toolchain
  # use GCC as the linker by commenting out both lines above and then
  # uncommenting the three lines below
  # "-C", "linker=arm-none-eabi-gcc",
  # "-C", "link-arg=-Wl,-Tlink.x",
  # "-C", "link-arg=-nostartfiles",
]

[build]
# Pick ONE of these compilation targets
# target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+
# target = "thumbv7m-none-eabi"    # Cortex-M3
# target = "thumbv7em-none-eabi"   # Cortex-M4 and Cortex-M7 (no FPU)
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)

```

build.rs

```
use std::env;
use std::fs::File;
use std::io::Write;
use std::path::PathBuf;

fn main() {
    // Put the linker script somewhere the linker can find it
    let out = &PathBuf::from(env::var_os("OUT_DIR").unwrap());
    File::create(out.join("memory.x"))
        .unwrap()
        .write_all(include_bytes!("memory.x"))
        .unwrap();
    println!("cargo:rustc-link-search={}", out.display());

    // Only re-run the build script when memory.x is changed,
    // instead of when any part of the source code changes.
    println!("cargo:rerun-if-changed=memory.x");
}
```

memory.x

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* TODO Adjust these memory regions to match your device memory layout */
  /* These values correspond to the LM3S6965, one of the few devices QEMU can emulate */
  FLASH : ORIGIN = 0x08000000, LENGTH = 64K
  RAM : ORIGIN = 0x20000000, LENGTH = 16K
}

/* This is where the call stack will be allocated. */
/* The stack is of the full descending type. */
/* You may want to use this variable to locate the call stack and static
   variables in different memory regions. Below is shown the default value */
/* _stack_start = ORIGIN(RAM) + LENGTH(RAM); */

/* You can use this symbol to customize the location of the .text section */
/* If omitted the .text section will be placed right after the .vector_table
   section */
/* This is required only on microcontrollers that store some configuration right
   after the vector table */
/* _stext = ORIGIN(FLASH) + 0x400; */

```

src/main.rs

```
#![no_std]
#![no_main]

#[macro_use]
extern crate cortex_m;
extern crate cortex_m_rt;
extern crate cortex_m_semihosting;

// pick a panicking behavior
// extern crate panic_halt; // you can put a breakpoint on `rust_begin_unwind` to catch panics
// extern crate panic_abort; // requires nightly
// extern crate panic_itm; // logs messages over ITM; requires ITM support
extern crate panic_semihosting; // logs messages to the host stderr; requires a debugger
                                // extern crate stm32f302;

use core::fmt::Write;
use cortex_m_semihosting::hio;

use cortex_m_rt::entry;

#[entry]
fn entry() -> ! {
    // iprintln!("Hello, world!");
    print();

    /*
    let peripherals = stm32f302::Peripherals::take().unwrap();
    let gpiob = peripherals.GPIOB;
    let rcc = peripherals.RCC;
    
    rcc.ahbenr.write(|w| w.iopben().bit(true));
    gpiob.moder.write(|w| unsafe { w.moder13().bits(0x01) });
    gpiob.odr.write(|w| unsafe { w.odr13().bit(true) });
    */

    loop {
        // your code goes here
    }
}

fn print() -> Result<(), core::fmt::Error> {
    let mut stdout = match hio::hstdout() {
        Ok(fd) => fd,
        Err(()) => return Err(core::fmt::Error),
    };

    write!(stdout, "Hello, world!")?;

    Ok(())
}
```

adamgreig, 2018-10-01:
I think the issue might be just that you're not actually using the stm32f3 crate in your code, so it's not getting pulled in and thus the interrupt vectors aren't being added.

Does adding `extern crate stm32f3;` to `main.rs` fix your issue?

adamgreig, 2018-10-07:
@sprhawk could you confirm if `extern crate stm32f3` in your `main.rs` fixes the issue?

adamgreig, 2018-10-24:
Closing the issue as I think it's resolved; shout if you're still having problems.

closed 2018-10-24 by adamgreig

sprhawk, 2018-10-25:
> @sprhawk could you confirm if `extern crate stm32f3` in your `main.rs` fixes the issue?

Sorry for late reply. Busy on doing other things lately.

I think you are right, it is being built now

Thanks