stm32-rs/stm32-rs #66: Can't build with 1.30-beta

thejpster, 2018-10-03:
I'm building with the 1.30-beta (as we're heading towards Stable Embedded in Rust 2018) and I get a few errors about the following being unsupported:

```
#![feature(const_fn)]
#![feature(try_from)]
#![cfg_attr(feature = "rt", feature(global_asm))]
#![cfg_attr(feature = "rt", feature(use_extern_macros))]
#![cfg_attr(feature = "rt", feature(used))]
```

Is there a roadmap for stable support in Rust 2018?

adamgreig, 2018-10-03:
I think those are actually all legacy directives which used to be generated by svd2rust but now are not, so we should actually be clear to just remove them from the `lib.rs` in each crate. I'll give it a go and push new versions if it works out.

adamgreig, 2018-10-03:
Yep, that fixes the issue, and builds work on beta and nightly. I'll try and close out the outstanding PRs and then cut another release.

adamgreig, 2018-10-07:
Fixed with v0.3.0, now on crates.io.

closed 2018-10-07 by adamgreig