stm32-rs/stm32-rs #67: Include does not work on peripherals

octronics, 2018-10-05:
To compose a peripheral from multiple common fields, I try to use the _include. But it does not documents the fields.

I declare the common fields on a separate file:
```
REG:
   FIELD:
      VALUE0: [0, "description"]
      VALUE1: [1, "description"]
```
Then I declare the peripheral on its file:
```
PERIPH:
  _include:
     - ../peripherals/periph/common_field.yaml
```
I had to use `../peripherals/periph/` path otherwise `svdpatch` complain it doesn't find the `stm32-rs/devices/common_field.yaml` file.

The actual problem is that the values are not added to the patched svd file.
Do I need to done something different? If not can you help me to made this functionality working.

Thanks.

octronics, 2018-10-08:
The top level `_include` works. But the one I have problem with is the `_include` inside a PERIPH block. There are none is the repositories. The only place I found was on the README. I can't use the top level because I have common fields between LPUART and USART and because some device has both but with different fields. The second snippet show what I need on the peripheral files. if you want I can provide you a work in progress branch where the problem occurs.

adamgreig, 2018-10-24:
I'm in the process of fixing this properly, but in the meantime... if you change the order of includes so the peripheral file is loaded first, then the common_patches second, it should work as expected. The part of including that updates dictionaries skips `_include` keys in dictionaries, so if the `USART*` dict is first loaded from a `common_patch`, it never gets subsequently updated with the required includes.

I'll hopefully have the fixed version up soon though.

adamgreig, 2018-10-24:
I've pushed an updated svdpatch which handles this properly. #75 will need some work to get it doing the right thing now though.

closed 2018-10-24 by adamgreig