stm32-rs/stm32-rs #69: STM32F4 family ETH_MACIIDR RM0090 discrepancy

therealprof, 2018-10-07:
peripherals/eth/eth_mac.yaml says:

```
  MACMIIDR:
    TD: [0, 0xFFFF]
```

However the RM0090 says the field is support to be called `MD`. I guess the reason it is called `TD` is that some other series (or special sibling) call it `TD` in the documentation but I'm wondering how we want to treat such discrepancies...

adamgreig, 2018-10-07:
Another ST classic.

The SVD file ST provide for the F405 -- and F469, F429, F427, F407, F7x7, F7x9, F7x6, and F7x5 -- all call it TD. The SVD files for the F101, 103, 107, 215, and 217 all call it MD. All the RMs I checked call it MD instead.

I think the solution is to rename TD to MD in the relevant SVD files, and update the `eth_mac.yaml` to say MD as well.

adamgreig, 2018-10-07:
```
$ rg -A8 MACMIIDR *.svd | rg TD
stm32f7x5.svd-              <name>TD</name>
stm32f7x6.svd-              <name>TD</name>
stm32f7x9.svd-              <name>TD</name>
stm32f7x7.svd-              <name>TD</name>
stm32f405.svd-              <name>TD</name>
stm32f407.svd-              <name>TD</name>
stm32f427.svd-              <name>TD</name>
stm32f429.svd-              <name>TD</name>
stm32f469.svd-              <name>TD</name>
$ rg -A8 MACMIIDR *.svd | rg MD
stm32f101.svd-              <name>MD</name>
stm32f103.svd-              <name>MD</name>
stm32f107.svd-              <name>MD</name>
stm32f215.svd-              <name>MD</name>
stm32f217.svd-              <name>MD</name>
```

therealprof, 2018-10-07:
Why do the F101/F103/F215/F405 have descriptions for ethernet at all?

adamgreig, 2018-10-07:
> Another ST classic.

adamgreig, 2018-10-07:
At the moment we can't remove or delete peripherals/registers/fields from the SVD in a yaml file, but it's becoming increasingly clear we need to for things like this. I think the more pressing thing would be changing TD to MD and releasing 0.3.2 really soon (it's technically a breaking change but if 0.3.1 has been out for an hour I wouldn't feel too terrible pushing it into that...).

Removing the ethernet peripheral from F101 F103 F215 F405 is slightly lower priority I think, but should still happen.

therealprof, 2018-10-07:
Oh, don't rush it. There's more coming. ;)

PR #71 (2018-10-07): Rename ETH_MACIIDR field TD to MD to align with RM0090
Fixes #69
commit f1e7d17f (2018-10-07): Rename ETH_MACIIDR field TD to MD to align with RM0090

Fixes #69
closed 2018-10-07 by adamgreig