stm32-rs/stm32-rs #72: Consider cross-referencing SVDs to C header files

adamgreig, 2018-10-07:
modm have a lovely repository of slightly-patched C header files extracted from CubeMX [here](https://github.com/modm-io/cmsis-header-stm32). It's not beyond the realm of imagination that we could automatically compare the C header definitions to the SVDs, and catch incorrectly named, missing, and surplus entries.

cs2dsb, 2018-12-31:
Have you seen the XML files in STM32CubeMX? [Someone has uploaded a copy here](https://github.com/esden/stm32cube-database)

They seem to specify every possible variant of each peripheral (db/mcu/ip/*.xml) and which of them exist on which chip (db/mcu/*.xml).

The peripheral files seem to have a complete list of enumerated values - I've only spot checked but certainly for the MCU I was using the XML was more complete than the SVD or the header files.

Reverse engineering it and doing something useful with them might be a bit of a task but if nothing else it's easier to copy enumerated values from there than the dodgy formatted pdf reference docs. For example:
```
<RefParameter Comment="PLL Mul" DefaultValue="RCC_PLL_MUL2" Name="PLLMUL" Type="list" Visible="false">
  <PossibleValue Comment="2" Value="RCC_PLL_MUL2"/>
...
```

cs2dsb, 2019-03-08:
I've had a bash at getting something useful out of the CubeMx db xml files. I focussed on getting the interrupts out because the interrupts for the micro I'm using are a mess and managed to generate a complete list:
```
WWDG:
  _add:
    _interrupts:
      WWDG:
        description: Window watchdog interrupt
        value: 0
EXTI:
  _add:
    _interrupts:
      PVD:
        description: PVD interrupt through EXTI line 16
        value: 1
      EXTI0:
        description: EXTI line0 interrupt
        value: 6
      EXTI1:
        description: EXTI line1 interrupt
        value: 7
      EXTI2:
        description: EXTI line2 interrupt
        value: 8
      EXTI3:
        description: EXTI line3 interrupt
        value: 9
      EXTI4:
        description: EXTI line4 interrupt
        value: 10
      EXTI9_5:
        description: EXTI line[9:5] interrupts
        value: 23
      EXTI15_10:
        description: EXTI line[15:10] interrupts
        value: 40
RTC:
  _add:
    _interrupts:
      TAMP_STAMP:
        description: RTC tamper and timestamp interrupts through EXTI line 21
        value: 2
      RTC_WKUP:
        description: RTC wake-up interrupt through EXTI line 22
        value: 3
      RTC_Alarm:
        description: RTC alarms A and B interrupt through EXTI line 17
        value: 41
FLASH:
  _add:
    _interrupts:
      FLASH:
        description: Flash global interrupt
        value: 4
RCC:
  _add:
    _interrupts:
      RCC:
        description: RCC global interrupt
        value: 5
DMA1:
  _add:
    _interrupts:
      DMA1_Stream0:
        description: DMA1 stream0 global interrupt
        value: 11
      DMA1_Stream1:
        description: DMA1 stream1 global interrupt
        value: 12
      DMA1_Stream2:
        description: DMA1 stream2 global interrupt
[... 184 lines cut ...]
FPU:
  _add:
    _interrupts:
      FPU:
        description: FPU global interrupt
        value: 81
SPI4:
  _add:
    _interrupts:
      SPI4:
        description: SPI4 global interrupt
        value: 84
SPI5:
  _add:
    _interrupts:
      SPI5:
        description: SPI5 global interrupt
        value: 85
```

I'm currently reading the mcu xml file to get a list of the IP it has and the NVIC IP xml file to get the interrupts. Using the CubeMx executable as reference I managed to extract the logic they use to get the peripheral names from the interrupt list. 
The C header defines for the values are directly constructed from info in the NVIC xml so it's fairly straightforward to extract the values. I've currently got a little bash file that finds all the defines for a given symbol in the headers for all the different devices f0-7 and plops out the values. As it turns out, the f0 often has different interrupt values but all the other devices are the same. In theory the same can be done with peripheral and register addresses - these may be harder as they can be compound defines - I was wondering about generating little c programs that just print the define I'm looking for...
I'm also using the svd parser from svd2rust to cross reference the peripheral names so it'll only generate output if the peripheral names it's found in the NVIC IP xml match peripherals in the svd file.

I thought I'd comment to say that at least in principle there is some useful data to be extracted but also to see if anyone had any ideas where we should go with it? Currently it's a janky command line tool that can be coerced into giving me the above output which can become a yaml file in this repo once #163 is resolved but in theory it could generate an entirely new SVD. Alternatively it could just be used to fact check the SVD files and print out a big list of differences which we could patch until the tool says the SVD is right.

salkinium, 2021-06-22:
> modm have a lovely repository of slightly-patched C header files extracted from CubeMX

Thanks, though ST has since published [the header files themselves here](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer#stm32cube-cmsis) and I was able to get most patches fixed. They are very responsive to feedback.

> It's not beyond the realm of imagination that we could automatically compare the C header definitions to the SVDs, and catch incorrectly named, missing, and surplus entries.

Yes, I've had the same idea and I've [written an extractor for that using GCC to resolve the macros](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_header.py#L127-L293). Unfortunately I didn't get to the point of dumping it in SVD files, only [in yet another XML format](https://github.com/salkinium/stm32-memory). Are you still interested in converting the C header files into SVD?

Btw: Here's a [short script to very easily get the latest CubeMX database](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/raw-data-extractor/extract-stm32.py). (yes, ST serves it over HTTP).

adamgreig, 2021-06-23:
Oh neat, thanks for the update! At the moment I'm less concerned about getting the C data into SVD and more getting both the SVD and the C data into a format we can compare (even just in-memory) to try and identify discrepancies that indicate errors in the SVDs that should be fixed. It might also be interesting to directly turn the (presumably less buggy) C headers directly into a Rust library without going through an SVD at all, depending on how much extra metadata we can get at the same time.