stm32-rs/stm32-rs #76: Allow regex on `_modify` block

octronics, 2018-10-21:
A use case:
On DBG peripheral registers some field are prefixed with the peripheral name (`DBG_`). I want to remove the prefix to provide better name. Writing all the fields that need patching by hand is not an option. They are numerous, not always present (not a problem as _modify don't bell on absent fields).

Allowing to use regex matches will allow to express this requirements as follow:
```yaml
DBG:
  "APB1_FZ":
    _modify:
     "DBG_(.*)":
       name: "$1"
```
It can be used elsewhere specially on H7 family that has so much registers to rename.
What are your feelings about this idea?

octronics, 2018-10-22:
Thanks. I'll wait for your implementation. Then I would used it all over the place. I am currently working on a patch series for documenting USB peripheral and this functionality would be very welcome.

adamgreig, 2018-10-29:
I'm still looking into this but won't have anything done before Wednesday. One sticking point is that it's not actually backwards compatible with fnmatch at all, so I think it might have be optional per-match rather than the default (I don't really want to try changing all the existing ones).

burrbull, 2025-05-27:
I'm closing this issue as we already have "_strip" rule for this issue.
But you could discuss your ideas in https://github.com/rust-embedded/svdtools/issues/49 of svdtools repo.

closed 2025-05-27 by burrbull