stm32-rs/stm32-rs #88: Warnings from svd2rust

birkenfeld, 2018-11-05:
svd2rust does emit a few warnings for most crates, e.g.
```
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr2_input")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr2_input")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr2_input")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr2_input")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("ccmr1_input")) overlaps with another register block at offset 24. Ignoring.
WARNING Some(Ident("fs_grxstsr_host")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("fs_gnptxfsiz_host")) overlaps with another register block at offset 40. Ignoring.
WARNING Some(Ident("otg_hs_grxstsr_peripheral")) overlaps with another register block at offset 28. Ignoring.
WARNING Some(Ident("otg_hs_grxstsp_peripheral")) overlaps with another register block at offset 32. Ignoring.
WARNING Some(Ident("otg_hs_tx0fsiz_peripheral")) overlaps with another register block at offset 40. Ignoring.
```
Should we try to fix these via patches, or is it impossible?

burrbull, 2018-11-05:
https://github.com/rust-embedded/svd2rust/pull/192

birkenfeld, 2018-11-05:
Got it, thanks.

closed 2018-11-05 by birkenfeld