stm32-rs/stm32-rs #91: STM32L4x2: RCC_APB1ENR1.USBFSEN renamed to usbf?

nickray, 2018-11-20:
Maybe I'm missing something, but why is the `USBFSEN` bit named `usbf`? https://stm32l4.stalder.io/stm32l4/stm32l4x2/rcc/apb1enr1/struct.W.html#method.usbf

Is this possibly a sed gone wrong?

birkenfeld, 2018-11-20:
Unfortunately, the SVD files supplied by ST are full of errors, omissions and inconsistencies. This strange name is one of them.

Since you're likely to run into a few more on the way, [helping out with a SVD patch](https://github.com/adamgreig/stm32-rs#updating-existing-devicesperipherals) is not hard :smile:

nickray, 2018-11-20:
Like [this](https://github.com/adamgreig/stm32-rs/pull/92)?

Thanks for the pointer, great to be able to contribute back :)

closed 2018-11-20 by adamgreig