stm32-rs/stm32-rs #104: Adapt SVD patches from stm32f103xx

mvirkkunen, 2018-12-10:
There are some valuable SVD patches in the stm32f103xx crate that could be adapted to the stm32f1 crate. Among other things they add enumerated values for many fields which currently only have a `bits()` method in stm32f1.

The patches are a plain patch file against the SVD XML files, so converting them to the format used in this project requires some work.

https://github.com/japaric/stm32f103xx/blob/ca6a494e497a2ab7428a81a60d4fd1035d906977/STM32F103xx.patch

therealprof, 2018-12-12:
@mvirkkunen I don't actually mind the bit(s) stuff, it's actually easier to see what is going on than when hiding them behind made up names by looking at the datasheet.

Of course the USB stuff needs fixing, that's just completely bogus.

burrbull, 2019-07-29:
This issue can be closed.

adamgreig, 2019-07-29:
:+1:, thanks for all your hard work on the F1 patches!

closed 2019-07-29 by adamgreig