stm32-rs/stm32-rs #143: stm32f407 `Interrupt` is neither `Clone` nor `Copy`

cbiffle, 2019-01-25:
I notice that the `Interrupt` enumeration for stm32f407 (at least) has move and identity semantics. Given that the `cortex_m` NVIC driver takes interrupt enumerations by value to do things like enable/disable interrupts, it seems like `Interrupt` ought to be `Copy`.

However, it's sufficiently weird that it isn't already that I thought I'd ask if there was some intent I had missed before sending a patch.

(FWIW I also strongly prefer such enumerations deriving `Debug`.)

cbiffle, 2019-01-25:
I see that this is the result of a tool you're using. I've filed [a bug against it](https://github.com/rust-embedded/svd2rust/issues/272).

cbiffle, 2019-01-26:
Okay, I've pushed a PR against svd2rust. Hasn't been accepted yet.

I attempted to patch stm32-rs's svd2rust dependency locally to unblock myself (using cargo's `[patch]` section), but this doesn't actually work -- since stm32-rs's build system is outside of Cargo and requires a bunch of system dependencies, it looks like you're going to have to integrate the new svd2rust version upstream.

So I'm reopening this to track the need for that bump once the PR is accepted.

adamgreig, 2020-01-21:
I believe this is resolved now.

closed 2020-01-21 by adamgreig