stm32-rs/stm32-rs #151: Discrepancy in available register write operations?

tegenfase, 2019-02-10:
Hey! This might be just my unfamiliarity with the Rust language (got started on it yesterday). 
I succesfully compiled a program for my STM32F0Discovery (stm32f051) using the "rust embedded" documentation, but so far it only covers returning strings via gdb. Which is rather unexciting... I want to blink a led!

I eventually got PC9 to a high state. But the process was very unintuitive mainly because I can not use the same syntax for the MODER and ODR registers. Is this a bepkac a bug or expected behaviour?
`
    let mut peripherals = stm32f0x1::Peripherals::take().unwrap();
    let rcc = peripherals.RCC;

    rcc.ahbenr.modify(|_r,w| w.iopcen().bit(true));

    let gpioc = &peripherals.GPIOC;
    gpioc.moder.modify(|_,w| w.moder9().bits(1));
    gpioc.odr.modify(|_,w| w.odr9().set_bit());
`

tegenfase, 2019-02-10:
I should probably clarify the issue a bit more:
`` w.odr9().set_bit() `` 
is valid, but
`` w.moder9().set_bit()  ``
is not valid.

burrbull, 2019-02-10:
set_bit / clear_bit available only for fields with 1 bit width.

This is what you need:
https://docs.rs/stm32f0xx/0.3.0/stm32f0xx/gpioa/moder/struct._Moder9W.html

birkenfeld, 2019-02-10:
The thing here is that there are four modes, so "set the mode bit" doesn't make much sense. As @burrbull said, there are dedicated methods to set each of the mode constants, and avoid inscrutable numeric constants.

Also, for setting GPIO pins high/low you'll normally want to use the `BSRR` registers, to avoid the memory read that `modify` has to perform.

tegenfase, 2019-02-10:
Yes I didn't realize the MODER register has 2 bits per pin (pebkac it is). I had not found this piece of documentation yet maybe it would be nice to link to it in the README.md as it's much more useful to newcomers?
Also thanks for the tip @birkenfeld

burrbull, 2019-02-10:
You also can use HALs.
For example:
[LED blinky](https://github.com/stm32-rs/stm32f0xx-hal/blob/master/examples/blinky.rs)

adamgreig, 2019-02-17:
I think the actual question in this issue has been answered. The crate pages do link to the docs but the crate READMEs don't and certainly could; I'll open a separate issue for that.

closed 2019-02-17 by adamgreig