stm32-rs/stm32-rs #158: Fresh project, "Could not compile `cortex-m-rt`. LLVM ERROR: Global variable 'HardFault_' has an invalid section specifier '.HardFault.default': mach-o section specifier requires a segment whose length is between 1 and 16 characters."

theronic, 2019-02-26:
Trying to compile a skeleton project with only this dependency on OSX throws compilation error about cortex-m-rt despite specifying latest version of cortex-m-rt. Do I need to cross-compile?

## Cargo.toml:

```
[package]
name = "stm-test"
version = "0.1.0"
authors = ["My name"]
edition = "2018"

[dependencies]
cortex-m-rt = "0.6.7"

[dependencies.stm32f4]
version = "0.6.0"
features = ["stm32f446", "rt"]

[dependencies.stm32f4]
version = "0.6.0"
features = ["stm32f446", "rt"]
```

~/src/main.rs:

```
extern crate stm32f4;
use stm32f4::stm32f405;

fn main() {
	let mut peripherals = stm32f405::Peripherals::take().unwrap();
    println!("Hello, world!");
}
```

When I try to compile:

```
cargo  new stm-test
cd stm-test
# edit files here
cargo run
    Updating crates.io index
  Downloaded stm32f4 v0.6.0
  Downloaded cortex-m-rt v0.6.7
  Downloaded cortex-m-rt-macros v0.1.5
   Compiling proc-macro2 v0.4.27
   Compiling semver-parser v0.7.0
   Compiling unicode-xid v0.1.0
   Compiling rand_core v0.4.0
   Compiling cortex-m-rt v0.6.7
   Compiling vcell v0.1.0
   Compiling cortex-m v0.5.8
   Compiling stm32f4 v0.6.0
   Compiling r0 v0.2.2
   Compiling aligned v0.2.0
   Compiling volatile-register v0.2.0
   Compiling rand_core v0.3.1
   Compiling rand v0.5.6
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.4
   Compiling quote v0.6.11
   Compiling syn v0.15.26
   Compiling cortex-m-rt-macros v0.1.5
LLVM ERROR: Global variable 'HardFault_' has an invalid section specifier '.HardFault.default': mach-o section specifier requires a segment whose length is between 1 and 16 characters.
error: Could not compile `cortex-m-rt`.
```

birkenfeld, 2019-02-26:
Are you setting a proper build target? It looks like rustc is trying to produce MacOS binaries.

For embedded you need a few more files than usual, look at the cortex-m-quickstart repository for a good example.

theronic, 2019-02-26:
Thanks! Got it working - will update and close with details. Mostly it was target. It would be great if the README mentioned the cross compile and linked to that quickstart (maybe I just missed it).

adamgreig, 2019-03-08:
I'm closing this for now as I think it's resolved but please do update with details later if you want to, @theronic.

closed 2019-03-08 by adamgreig