stm32-rs/stm32-rs #184: Deal with conflicting cortex_m versions

therealprof, 2019-04-12:
When using a different cortex_m version than the one declared by stm32-rs the Rust compiler encounters a conflict in enum types

```
   = note: expected type `cortex_m::peripheral::syst::SystClkSource` (enum `cortex_m::peripheral::syst::SystClkSource`)
              found type `cortex_m::peripheral::syst::SystClkSource` (enum `cortex_m::peripheral::syst::SystClkSource`)
```

but also in static symbols:

`warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!`

Maybe those resources should be re-exported so the application can simply use them instead of trying to match up with the version in use by stm32-rs?

adamgreig, 2019-04-12:
Is this more of an svd2rust issue?

therealprof, 2019-04-12:
Good question. Might make sense to solve it there but I've no idea how.

adamgreig, 2019-04-12:
As of the next stm32-rs release we'll depend on ">=0.5.8&<0.7" which should let stm32-rs just use the same cortex_m as the target crate, hopefully resolving this problem.

therealprof, 2019-04-12:
I will try this right now.

therealprof, 2019-04-12:
Just as I suspected: No change, it still picks up two different versions.

adamgreig, 2019-04-12:
Huh, that doesn't sound right. Why is it bringing in two versions, and which versions? If the stm32 crate is `>=0.5.8&<0.7` and another crate is `0.6.0` then AIUI cargo should just use 0.6.0...

therealprof, 2019-04-12:
```
tm32f4-smoltcp-demo v0.1.0 (/Users/egger/OSS/stm32f4-smoltcp-demo)
├── cortex-m v0.6.0
│   ├── aligned v0.3.1
│   │   └── as-slice v0.1.0
│   │       ├── generic-array v0.12.0
│   │       │   └── typenum v1.10.0
│   │       └── stable_deref_trait v1.1.1
│   ├── bare-metal v0.2.4
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   └── volatile-register v0.2.0
│       └── vcell v0.1.0
├── cortex-m-rt v0.6.8
│   ├── cortex-m-rt-macros v0.1.5
│   │   ├── proc-macro2 v0.4.20
│   │   │   └── unicode-xid v0.1.0
│   │   ├── quote v0.6.9
│   │   │   └── proc-macro2 v0.4.20 (*)
│   │   ├── rand v0.5.5
│   │   │   └── rand_core v0.2.2
│   │   │       └── rand_core v0.3.0
│   │   └── syn v0.15.18
│   │       ├── proc-macro2 v0.4.20 (*)
│   │       ├── quote v0.6.9 (*)
│   │       └── unicode-xid v0.1.0 (*)
│   └── r0 v0.2.2
├── panic-halt v0.2.0
├── smoltcp v0.5.0
│   ├── bitflags v1.0.4
│   ├── byteorder v1.2.7
│   └── managed v0.7.1
├── stm32f4-smoltcp v0.1.0 (/Users/egger/OSS/stm32f4-smoltcp)
│   ├── cortex-m v0.6.0 (*)
│   ├── smoltcp v0.5.0 (*)
│   └── stm32f4xx-hal v0.4.0
│       ├── bare-metal v0.2.4 (*)
│       ├── cast v0.2.2
│       ├── cortex-m v0.6.0 (*)
│       ├── cortex-m-rt v0.6.8 (*)
│       ├── embedded-hal v0.2.2
│       │   ├── nb v0.1.1
│       │   └── void v1.0.2
│       ├── nb v0.1.1 (*)
│       ├── stm32f4 v0.6.0
│       │   ├── bare-metal v0.2.4 (*)
│       │   ├── cortex-m v0.5.8
│       │   │   ├── aligned v0.2.0
│       │   │   ├── bare-metal v0.2.4 (*)
│       │   │   └── volatile-register v0.2.0 (*)
│       │   ├── cortex-m-rt v0.6.8 (*)
│       │   └── vcell v0.1.0 (*)
│       └── void v1.0.2 (*)
└── stm32f4xx-hal v0.4.0 (*)
```

After removing the line causing the compiler error, I'll get this linker error:
```
  = note: rust-lld: error: duplicate symbol: CORE_PERIPHERALS
          >>> defined at mod.rs:153 (/Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.6.0/src/peripheral/mod.rs:153)
          >>>            cortex_m-3273e13c28219ec5.cortex_m.7wds9vvz-cgu.0.rcgu.o:(CORE_PERIPHERALS) in archive /Users/egger/OSS/stm32f4-smoltcp-demo/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-3273e13c28219ec5.rlib
          >>> defined at mod.rs:153 (/Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.8/src/peripheral/mod.rs:153)
          >>>            cortex_m-f1137fe84c50646a.cortex_m.1fscb29e-cgu.0.rcgu.o:(.bss.CORE_PERIPHERALS+0x0) in archive /Users/egger/OSS/stm32f4-smoltcp-demo/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-f1137fe84c50646a.rlib
```

tim-seoss, 2019-04-20:
> `│       ├── stm32f4 v0.6.0`

Doesn't that imply that you're just using the released version without the updated dependency, rather than a local "patch"?  Otherwise (after running `cargo update`) the local path to the patched version would appear like this:

``` $ cargo tree
owrt v0.1.0 (/home/tim/prog/rust/nucleo-f446re/owrt)
├── cortex-m v0.6.0
[...]
    ├── stm32f4 v0.6.0 (/home/tim/prog/rust/stm32-rs/stm32f4)
    │   ├── bare-metal v0.2.4 (*)
    │   ├── cortex-m v0.6.0 (*)
```
?

adamgreig, 2019-04-22:
Yes, I'm pretty sure this issue is just because current stm32-rs requires cortex-m 0.5, other dependencies might require 0.6, and so Cargo will build both and that deliberately fails. Once stm32-rs 0.7 is released (this week, once pending PRs are done), it depends on >=0.5.8&<0.7, which will allow using new or old cortex-m.

adamgreig, 2019-05-06:
I believe we've now solved this with cortex-m 0.5.10.

closed 2019-05-06 by adamgreig

little-arhat, 2021-01-24:
```
warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!

error: failed to load bc of "cortex_m-5c0c4694ee41d0fc.cortex_m.63gufsr0-cgu.0.rcgu.o":

error: aborting due to previous error; 1 warning emitted
```

cortex-m 0.6, loaded via this stm32f3 conflicts with cortex-m 0.7 used by every other crate. Is there a way to fix that?

```
├── alt-stm32f30x-hal v0.31.0
│   ├── bitrate v0.1.1
│   ├── bobbin-bits v0.1.1
│   ├── cast v0.2.3
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── cortex-m v0.7.0
│   │   ├── bare-metal v0.2.5
│   │   │   [build-dependencies]
│   │   │   └── rustc_version v0.2.3 (*)
│   │   ├── bitfield v0.13.2
│   │   ├── embedded-hal v0.2.4
│   │   │   ├── nb v0.1.3
│   │   │   │   └── nb v1.0.0
│   │   │   └── void v1.0.2
│   │   └── volatile-register v0.2.0
│   │       └── vcell v0.1.3
│   ├── embedded-hal v0.2.4 (*)
│   ├── nb v1.0.0
│   ├── stm32f3 v0.12.1
│   │   ├── bare-metal v0.2.5 (*)
│   │   ├── cortex-m v0.6.0
│   │   │   ├── aligned v0.3.4
│   │   │   │   └── as-slice v0.1.4
│   │   │   │       ├── generic-array v0.12.3
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       ├── generic-array v0.13.2
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       ├── generic-array v0.14.4
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       │   [build-dependencies]
│   │   │   │       │   └── version_check v0.9.2
│   │   │   │       └── stable_deref_trait v1.2.0
│   │   │   ├── bare-metal v0.2.5 (*)
│   │   │   └── volatile-register v0.2.0 (*)
│   │   ├── cortex-m-rt v0.6.13
│   │   │   ├── cortex-m-rt-macros v0.1.8 (proc-macro)
│   │   │   │   ├── proc-macro2 v1.0.24
│   │   │   │   │   └── unicode-xid v0.2.1
│   │   │   │   ├── quote v1.0.8
│   │   │   │   │   └── proc-macro2 v1.0.24 (*)
│   │   │   │   └── syn v1.0.59
│   │   │   │       ├── proc-macro2 v1.0.24 (*)
│   │   │   │       ├── quote v1.0.8 (*)
│   │   │   │       └── unicode-xid v0.2.1
│   │   │   └── r0 v0.2.2
│   │   └── vcell v0.1.3
│   └── void v1.0.2
├── asm-delay v0.9.0
│   ├── bitrate v0.1.1
│   ├── cortex-m v0.7.0 (*)
│   └── embedded-hal v0.2.4 (*)
├── cortex-m v0.7.0 (*)
├── cortex-m-rt v0.6.13 (*)
├── embedded-hal v0.2.4 (*)
├── nb v0.1.3 (*)
├── panic-abort v0.3.2
└── stm32f3 v0.12.1 (*)
```