stm32-rs/stm32-rs #188: Proposal: minor releases

burrbull, 2019-04-22:
Could we make frequent minor releases (0.7.1, etc.) from unbreakable changes, like adding enumeratedValues, documentation or missing interrupts? This would make hal development cycle easier.
For example once a month minor release and once in 3 or 4 months major release.

adamgreig, 2019-04-22:
Yep, in principle I'm very happy to do this, even weekly if there have been a couple of PRs. The main problem is right now so many PRs are breaking changes and we don't have a good way of tracking which are breaking and which are not.

adamgreig, 2020-01-21:
I've now set up https://github.com/stm32-rs/stm32-rs-nightlies which automatically builds when new commits hit master, does this mitigate this issue? The git dependencies are usable for development/local builds but wouldn't work for HALs published on crates.io, those would have to wait for a new version.

I found doing actual minor releases quite hard as most changes were breaking.

closed 2025-05-27 by burrbull