stm32-rs/stm32-rs #191: stm32h7 ethernet naming

jordens, 2019-04-22:
(Except for some registers in `Ethernet_MAC` and the registers in `Ethernet_MMC`):

The STM32H7 ethernet peripherals and registers are currently names as `Ethernet_$X.$X$Y` (where e.g. `$X=DMA` and `$Y=DSR`). The ref manual typically names them `ETH_$X$Y`.

 @ehntoo Does it make sense to remove the `$X.$X` duplication and name that stuff `ETH_$X.$Y` (stripping the `$X` from the register name) or `ETH.$X$Y` (merging the ethernet peripherals)?

ehntoo, 2019-04-26:
Due to some career changes I haven't had any real opportunity to actually sit down and use the H7 Ethernet peripheral. As I recall, there was quite some variability in the way the reference manual referred to things. If there's a clean way to unify things, especially one that would stay consistent with other parts, I think it could well be worth pursuing. That said, I wouldn't place too much weight on my opinions here anyway, I've just been fiddling with embedded rust at odd hours in evenings and on weekends..

jordens, 2019-05-05:
The newer SVD has the MMC merged into the MAC and the MTL starts at `0x40028c00`.

PR #208 (2019-05-06): stm32h7 svd v1.4
## Major SVD changes:

* Several other devices have been added. Among the changes is the split of `stm32h7x3` into `stm32h743x`, `stm32h753x`, `stm32h7a3x` (??), `stm32h7b3x` (??).
* The new devices don't vary that much from `stm32h7x3` and the latter still seems to be a good superset. There is some RAM/ECC controller and Accelerator in the h7x7/h7x5 CM4/CM7 devices.
* Most of the upstream changes in the SVD are compatible with the current patches.
* There would be fallout w.r.t. RCC (if the pending #206 gets merged)
* and with Ethernet (added MAC, MTL, MMC peripherals upstrea [... cut]
closed 2019-05-06 by adamgreig