stm32-rs/stm32-rs #196: Easy access to patched SVDs

nickray, 2019-04-28:
In the nRF world, PACs contain the SVD (e.g. [nrf52840.svd](https://github.com/nrf-rs/nrf52840-pac/blob/master/nrf52840.svd)). This is nice, for instance [Cortex-Debug](https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/) allows importing an `svdFile` to offer a peripheral registers view. Can publish the SVDs somewhere?

adamgreig, 2019-04-28:
Good idea! I'd kept them out of git as they're big and keep having huge changes (one line change in the YAML can cause a large change in all 50 SVDs), but I already have stm32-rs built on my web server regularly to generate the web page, so I'll extend that to also include the latest patched SVDs.

adamgreig, 2019-04-28:
Seems to be working: https://stm32.agg.io/rs/

PR #198 (2019-04-28): Include links to patched SVDs in HTML output
Closes #196
closed 2019-04-28 by adamgreig