stm32-rs/stm32-rs #207: stm32h7 ethernet dma pointer width

richardeoin, 2019-05-03:
Fields `TDELSA`, `TDESLA`, `TDT`, `RDT` are 32-bit pointers, but are defined in the SVD and RM as [31:2] with 30 bit width. The two LSBs of the register are described as ready-only.

The equivalent registers in F4/F7 ethernet dma definitions have full 32-bit width. Should we change these fields to be the same? I'd say 32-bit width is closer to expected behaviour, but it is technically different to the RM.

Current workaround is just to write/read from the whole register
```rust
unsafe {
  eth_dma.dmactx_dlar.write(|w| {
    w.bits(&buf as *const _ as u32)
  });
}
```

jordens, 2019-05-06:
In both cases you would still have to manually guarantee alignment. That doesn't change.
The alternative would be:

```rust
let addr = &buf as *const _ as u32;
assert_eq!(addr & 0x3, 0);
eth_dma.dmactx_dlar.write(|w| w.dlar().bits(addr >> 2));
```

No `unsafe` but instead shifting gymnastics which makes it harder because they can be forgotten.
Personally I'd rather not have to see the shifting.

adamgreig, 2019-05-06:
This is the same issue as #195 raised about `BTABLE`, which also has alignment requirements which ST (well, Synopsys, who do both the USB and the Ethernet..) address by making a short and shifted field in the register.

On the one hand I agree that the shifting is annoying and prone to mistakes; on the other hand it seems like it would lead to surprising disagreements with the reference manual if we made the field wider ourselves. I sort of like just writing the address to the whole register instead of to the field (though you can move the `unsafe` to inside the closure which is neater I think, just use `.write(|w| unsafe {`).

Maybe there's an argument to be made that writing a memory address (which you can get safely) to a hardware register which is then going to start reading/writing that address should be unsafe anyway.

jordens, 2019-05-06:
I think I agree (with the `.write(|w| unsafe { w.bits(addr) })` being idiomatic in terms of `unsafe`ty and register interpretation).

richardeoin, 2019-05-10:
Thanks both for the discussion - this is really useful for my understanding of what is/isn't idiomatic. I'm happy that the unsafe write to the whole register is acceptable here, so closing issue.

closed 2019-05-10 by richardeoin