stm32-rs/stm32-rs #209: "Discussion" repository? HAL unification.

nickray, 2019-05-06:
Opening issue here as it seems the most general of the STM32 repositories.

I think it would be useful to have a separate repository, with issues and discussion pertaining to all things cross-STM32 in Rust.

My first use case would be to push to emulate (and improve + extend!) the nice things https://github.com/nrf-rs have:
- individual PACs based on patched SVDs that can be inspected (e.g. https://github.com/nickray/stm32l4x2-pac)
- constructors instead of extension traits in the HALs (c/o @hannobraun)
- workspace to organize all the things
- individual HALs as successive specialisations

The latter is probably the most important. Whereas they have `nrf52-hal-common` with three actual HALs, we probably need a multistep approach (stm32-hal-common -> stm32l4-hal-common -> stm32l4x2-hal-common -> stm32l432-hal?). Whereas they so far only seem to plug in the appropriate PAC in the generic HAL implementation, we may have various versions of USB peripherals, differing collections of clocks, etc. etc. An actual HAL should (imho) be for an actual chip, offering only (and hopefully all) functionality on that specific chip.

We have far too much duplication, I'm struggling to keep up (for my STM32L432 use case) with the ideas scattered in various implementations (https://github.com/stm32-rs/stm32l4xx-hal/issues/56, https://github.com/stm32-rs/stm32l4xx-hal/issues/57, https://github.com/stm32-rs/stm32l4xx-hal/pull/53, ...)

hannobraun, 2019-05-06:
> * workspace to organize all the things

I'd like to note that this is one thing that doesn't seem to work so well. I don't think we can build from the workspace over in nrf52-hal. Not sure why, but the [CI script](https://github.com/nrf-rs/nrf52-hal/blob/4c21ba5f0b5ef4012770ee76db6a9ae1884dc6cd/scripts/build.sh) builds all crates individually.

I know that we also had problems using a workspace over in [lpc8xx-hal](https://github.com/lpc-rs/lpc8xx-hal) because Cargo looks for the linker scripts at the workspace root instead of where they actually are. (lpc8xx-hal also does the other things suggested here, by the way)

adamgreig, 2019-05-06:
Do you think this (stm32-rs/stm32-rs) repo could work for discussion issues? As you've identified it's sort of central and appropriately named. If discussion issues get overwhelming here maybe we could then open a new repository just for tracking them.

> individual PACs based on patched SVDs that can be inspected

This is the point I'm least sold on. Once upon a time this is how stm32-rs worked but it quickly became untenable. The nrf-rs org has three independent PACs which (as I understand it) have infrequent, small, manual additions to the SVD files. We have 11 family PACs with about 50 devices in total, with regular and large churn in the SVDs and consequently svd2rust autogenerated code. The git repos would get huge and hard to use real quick, the commit diffs would be unreadable, etc. I can think of workarounds but none seem like a worthwhile cost/benefit tradeoff - though if people have good ideas here I'm open to hearing them. The best candidate I have so far is separate repositories that just host each released version of the PACs and are updated at release time, which exist just to let people browse the code. Another option is a repository per family crate which submodules stm32-rs to get at the SVDs and patching data then builds the crate out locally, but I really think the git churn would still be too high and it makes managing the SVD patches centrally much harder.

What I'd really like is another way to solve the problem, which I think is more around discovering the API than just being able to read heaps of autogenerated source on GitHub. The documentation on docs.rs is pretty good (and does include the autogenerated source code!) but build issues means it can't cover every device, and it's really slow to navigate because of the huge search index file that gets downloaded and parsed. Perhaps some svd2rust-specific tooling to generate really good API documentation would be the real win? We could then run that automatically at release and host it on a GitHub pages site, or the current [stm32-rs website](https://stm32.agg.io/rs).

Maybe just better documentation on how to view the generated code which already lives on your computer -- it's all there and available in your favourite text editor! We even specifically build with Form before releases to make the code in the crate download as readable as possible.

> constructors instead of extension traits in the HALs (c/o @hannobraun)

> individual HALs as successive specialisations

I don't have as much to do with the HAL implementations so I'll take a back seat here except to say I like the new constructors idea and am excited to see how it looks, and,

> An actual HAL should (imho) be for an actual chip

I don't agree; I think we should have one HAL that covers _every_ stm32 (or, realistically, compromise and have one HAL per family) rather than having 49 separate HALs we have to somehow manage and keep consistent. And 49 is only per SVD -- each SVD covers 2-5 chips, so we might be talking about literally hundreds of HALs. That number will only grow as ST continue to release more chips.

benpye, 2019-05-30:
I am currently implementing a HAL for the HT32F52342 - a Holtek Cortex-M0+ part.

I like the approach taken here compared to many of the other HALs, like ST Holtek have many different ARM Cortex microcontrollers, M0+ and M3, and multiple variants of the two. Whilst I'm initially targeting this specific chip I would like to be able to use similar tools to those used here to patch the SVD files. Holtek for example don't fill the description field, instead just repeating the name (doh) - and I have noticed a couple typos, even before executing any code!

It would be useful to have the various scripts used here for dealing with SVDs in a seperate repo so that I (and I am sure others) could benefit from the existing work - without needing to take a fork.

adamgreig, 2020-06-17:
While the scripts used for SVDs have moved to [svdtools](https://github.com/stm32-rs/svdtools), I've now also created the new [meta](https://github.com/stm32-rs/meta) repository for other coordination and discussions.

closed 2020-06-17 by adamgreig