stm32-rs/stm32-rs #226: Support for STM32F750

ondono, 2019-05-31:
I was thinking in getting started in embedded rust using a STM32F7508-DK board, that uses a STM32F50N8 Microcontroller.

I've seen there's apparently no support yet, not seeing it in the Device Coverage list either. I just wanted to make sure that there's no major roadblock that I'm not yet able to see, it's okay if it's just that nobody has started to work on it, I'm willing to do it.

Thank you!

adamgreig, 2019-05-31:
You're the second person to ask about F7x0 support recently! I'm on a work
trip at the moment so can't work on it yet, but it shouldn't be too bad if
you want to open a PR.

You'd need to update to the latest f7 SVD zip (in the svd/vendor folder)
and then add new 730 and 750 device yaml files. Ideally also include
appropriate patches and peripherals (see the other f7 device files for
inspiration).

Shout if you run into any issues or need help with anything.

ondono, 2019-06-06:
I went for it and I already have the board. I have quite a lot to figure out since I'm new to rust and I'm still trying to understand all the tooling and libraries to work with embedded devices.

I've found an issue though, and I'm not sure what's the procedure:
- I've loaded the latest .zip file from ST for the f7 family.
- The STM32F7X5.svd file is no longer present on the .zip file, I think it has been split into STM32F745.svd and STM32F765.svd
- I've also created STM32F745.yaml and STM32F765.yaml by just copy pasting the previous STM32F7X5.yaml file

This seems to at least let me build the modified stm32f7 crate, though I'm still modifying it away to add the 750. My main worry is that I have no way to test that I have not broken the 745 & 765 libs by accident since I have no way to test them on real hardware. What is the proper way forward?

ogoffart, 2019-06-15:
Hi @ondono   
I also have a STM32F7508-DK board and wanted to try rust on it. So thanks for your work.
I found your fork of this repository by clicking on your name and gave it a try

I get this error:

```
python3 scripts/svdpatch.py devices/stm32f7x9.yaml
[...]
  File "scripts/svdpatch.py", line 614, in process_field_enum
    .format(pname, rname, fspec))
__main__.MissingFieldError: Could not find FLASH:CR.MER
make: *** [Makefile:38: svd/stm32f7x9.svd.patched] Error 1
```

Did you also get that? 
Anyway since it is not related to the board i have, i just went ahead and tried to build an hello world. (a simple empty loop). And it worked.

It was hard to find a way to flash the board. Apparently one need a bootloader, so i used the C bootloader coming from the STM32Cube SDK, and used the flashing tool from there too.
openocd did not work for me.

So thanks for your work. Next step for me will be to get the screen working now.

> What is the proper way forward?

I'm not related to this crate in any way, but I'd say the best would be to make a pull request so it can be tested by the right people.

ondono, 2019-06-18:
> Did you also get that?

Yes that's one of the issues I found. As far as I can tell, the svds for various part numbers have been updated, so the yaml files are out of date and the patching procedures stops due to repeated keywords. I thought about doing a PR but I know that could easily lead to broken crates for other people even if tests are performed.

> Anyway since it is not related to the board i have, i just went ahead and tried to build an hello world. (a simple empty loop). And it worked.

I got pretty much to that point, got it to compile with a GPIO toggling but have not flashed it yet.

> It was hard to find a way to flash the board. Apparently one need a bootloader, so i used the C bootloader coming from the STM32Cube SDK, and used the flashing tool from there too.
> openocd did not work for me.

That sounds great, I got stuck trying to figure out how to flash the chip and then life got in the way...
Have you been able to confirm that the code is up and running by toggling a LED or something?

sayansarkarnitdgp, 2019-10-03:
Dear ondono,

Can you kindly explain C bootloader based flashing process in detail?

Thanks in advance 

Sayan Sarkar

chengsun, 2020-12-12:
I have gotten the STM32F7508-DK board to work with `stm32-rs` locally.

My pull request for adding the STM32F750 device to `stm32-rs` is pending https://github.com/stm32-rs/stm32-rs/pull/448 getting merged first.

Here's the OpenOCD config I used:

```
# OpenOCD configuration for the STM32F7508-DK development board
# requires OpenOCD 0.10.0+dev (more recent than 2020-01-03: http://openocd.zylin.com/#/c/4926/)

source [find interface/stlink.cfg]
source [find target/stm32f7x.cfg]
```

adamgreig, 2020-12-16:
Closed by #464.

closed 2020-12-16 by adamgreig