stm32-rs/stm32-rs #238: stm32f7 GPIO miss for everything in family

no111u3, 2019-06-18:
I start create hal for stm32f7xx mcu and found mistake - for everything controller in f7 family I can't see anything ports exclude GPIOA/B/D.
For example - https://docs.rs/crate/stm32f7/0.7.1/source/src/stm32f7x7/
has 3 ports but in fact there's has 7 ports.

adamgreig, 2019-06-18:
There are only 3 modules containing `RegisterBlock`, GPIOA, GPIOB, GPIOH, but several of them are instantiated multiple times to create the other GPIO ports, see [here](https://docs.rs/stm32f7/0.7.1/stm32f7/stm32f7x3/index.html#structs), e.g. [GPIOC](https://docs.rs/stm32f7/0.7.1/stm32f7/stm32f7x3/struct.GPIOC.html). It uses `GPIOH::RegisterBlock` as its contained type because all the GPIO registers are the same between C/D/E/F/G/H.

closed 2019-06-18 by adamgreig

no111u3, 2019-06-18:
Rust is strong typed languade - and same content isn't same type. I can't use your solution because I violence rust principles.

adamgreig, 2019-06-18:
This is the same for all the crates made with `svd2rust`, including all the other STM32 crates, but the HALs manage, usually using macros like you have. Each GPIO is still its own outer type, just the interior `RegisterBlock` is shared.

no111u3, 2019-06-19:
This is the different from japaric style and this unlogic because low sw level will be copy of hw layer. SVD have optimization because it's description - it has links for hierarhy navigation.