stm32-rs/stm32-rs #245: Unique device ID register (96 bits)

tkeksa, 2019-07-01:
Is the U_ID register supported? I can not find it.

adamgreig, 2019-07-01:
It's not in any of the SVD files ST publish, and no one's specifically added it. I think it would be an OK addition, but I've always just ended up reading it manually:

```rust
unsafe fn read_id() -> [u8; 12] {
    let id1: [u8; 4] = (*(0x1FFF_F7AC as *const u32)).to_le_bytes();
    let id2: [u8; 4] = (*(0x1FFF_F7B0 as *const u32)).to_le_bytes();
    let id3: [u8; 4] = (*(0x1FFF_F7B4 as *const u32)).to_le_bytes();
    [
        id1[0], id1[1], id1[2], id1[3],
        id2[0], id2[1], id2[2], id2[3],
        id3[0], id3[1], id3[2], id3[3],
    ]
}
```

(those are STM32F0x2 addresses, and will need adjusting as appropriate).

adamgreig, 2020-02-12:
We now have @disasm's [stm32-device-signature](https://github.com/stm32-rs/stm32-device-signature) crate, which allows accessing these IDs across families.

closed 2020-02-12 by adamgreig