stm32-rs/stm32-rs #251: stm32f7 documentation is missing stm32f7x6

hannobraun, 2019-07-11:
The [documentation of the `stm32f7` crate](https://docs.rs/stm32f7/0.7.1/stm32f7/) shows modules for `stm32f7x3` and `stm32f7x9`, but it's missing `stm32f7x6`.

adamgreig, 2019-07-20:
This is deliberate and set here: https://github.com/stm32-rs/stm32-rs/blob/master/scripts/makecrates.py#L28

Basically building too many features causes the docs.rs run to get OOM killed, and the F7 crate is especially huge. We already stall the whole docs.rs pipeline for >24h when doing a release. See #3 for more discussion and links to related docs.rs issues...

The hope is that the documentation we do show is close enough or otherwise users can generate documentation locally, which isn't ideal. Hopefully soon the document generation will be less resource intensive and docs.rs will build for only one target, both of which would help matters, and we could add more devices to the list.

closed 2019-07-20 by adamgreig

hannobraun, 2019-07-22:
Makes sense. Thank you for the explanation!