stm32-rs/stm32-rs #257: rcc_f0_f1_f3_common has two values with the same name

kyrias, 2019-07-16:
In `rcc_f0_f1_f3_common.yaml`, the `RCC.CFGR.PLLMUL` patch contains `Mul16` twice but with different values.  As a single `enumeratedValue` can't have multiple values, the second one ends up overriding the first.  This means that if the user would manually set `PLLMUL` to the first bit pattern, they would get back `PLLMULR::_Reserved` when reading back the value.

https://github.com/stm32-rs/stm32-rs/blob/f11c937464cf6782d0d19b27c26854ed2cdc0665/peripherals/rcc/rcc_f0_f1_f3_common.yaml#L21-L22

This is sort of an annoying situation.  I think that preferably the SVD format would support specifying multiple bit patterns for one enumeration, and then just using the first of them when writing.  But without changing the file format, the only work-around I can think of is naming the second one something like `Mul16_2`, and then seeing if `svd2rust` could be modified to detect this and merge them together for reads, so that the consumer doesn't have to care what the actual bit pattern was.

burrbull, 2019-07-16:
Futhermore f101-107 have differ `PLLMUL`:
[image]

kyrias, 2019-07-17:
Only F105 and F107, which make up the connectivity line of F1s, have that PLLMUL, whereas all other F1 series devices have the `PLLMUL` values currently specified.

burrbull, 2019-07-17:
Is #259 better?

adamgreig, 2019-07-24:
Closed by #259.

closed 2019-07-24 by adamgreig