stm32-rs/stm32-rs #261: `form` before publish docs

burrbull, 2019-07-19:
Docs.rs render anomaly slow.

little-arhat, 2019-07-19:
docs.rs also hangs sometimes when creating docs for this or dependent crates, see https://github.com/rust-lang/docs.rs/issues/335

adamgreig, 2019-07-19:
I'm not completely sure if `form` is responsible for the slowness - it increases the number of _input_ files that docs.rs has to process, but it results in the same number of _output_ HTML files which I think is more the issue. Anyway we run `form` on release specifically so that the published crates have a useful file structure for people checking the source code, so I'd rather keep that.

I'm happy to try anything to help reduce load on docs.rs and still end up with useful docs, hopefully their upcoming change to only build for one platform will make a big difference.

burrbull, 2019-07-19:
Yesterday was one `mod.rs` file for whole crate on docs.rs.
:smirk:

closed 2019-07-19 by burrbull

adamgreig, 2019-07-19:
> Yesterday was one mod.rs file for whole crate on docs.rs.

What crate? There haven't been any stm32-rs releases recently and any dependent crates shouldn't be building our docs.

burrbull, 2019-07-19:
f103. I have no idea what happened. Maybe I was looking at old version.