stm32-rs/stm32-rs #284: Split out `svdpatch.py`

nickray, 2019-08-17:
Hey @adamgreig 

I am about to fix a very long and very buggy NXP SVD file and have need for the SVD patching tool contained in this repository. To make this more comfortable, I copied `svdpatch.py`, added some packaging and CLI niceties, and published on PyPI:
- https://github.com/nickray/svdtools
- https://pypi.org/project/svdtools/

Thoughts? There's more in the scripts directory that could be pulled out.

Ideally, I'd like to hand over my repository and maintenance to someone else :)

adamgreig, 2019-08-17:
Ah, nice. When I started stm32-rs, svdpatch.py was one of a handful of small scripts but it's grown into a bit of a beast and really could do with being its own tool, more easily usable by other svd2rust projects, ideally with tests and other good software practice. Oh, and actual documentation for the YAML patching format! All of that ends up being a bit of work for low reward though, so I don't know when it will happen.

Perhaps we could move the svdtools repository to stm32-rs or even rust-embedded orgs for long-term maintenance, what do you think?

nickray, 2019-08-17:
Yeah, understood it's not glamorous at all... Maybe if it is independent, more `svd2rust` users will use (and improve it)?

Indeed my intent is to move this out of my hands again before it forks from your original (although it seems last change was 2019-04-29). Maybe you can figure out if it's trivial to adjust this `stm32-rs` repo to use my packaged version, to validate the approach? Also, what else would you want to extract to `svdtools`, if we proceed?

As to where to place it, I think you're a better judge than me - maybe to this new community org to decouple it from stm32 (assuming the rust-embedded org does not want to take on the workload, which would make sense since svd2rust isn't directly applicable to most SVDs, but understood everyone is busy already...).

On my side, I'm not 100% sure patching is the optimal approach, since I have one-ish SVD for the LPC55S6x chips, compared to your huge family of chips. Plus hopefully NXP will iterate a bit and fix their bugs, in which case I wonder how hard it will be to "rebase" the patches.

hannobraun, 2019-08-19:
As someone who's involved in all the vendor-specific orgs that I know of (stm32-rs, lpc-rs, nrf-rs), I think moving this into its own project (potentially under the WG) is a great step. We sorely need a standardized tool. lpc-rs alone is using two different shell-script-based approaches in its PAC repository.

I'd love to see the tool re-written in Rust though. We're part of the Rust community after all, so every single user of the tool would likely be comfortable contributing to a Rust project. Someone needs to do that though, and I'm not volunteering at this time :-)

(Just to clarify, I don't think there's a case for nrf-rs to start using this. The Nordic SVDs are so high-quality, I'm not aware of a single patch being applied to them.)

nickray, 2020-01-14:
@adamgreig shall I PR this repo to use svdtools? Latter is getting more users so I'd like to prevent a fork.

adamgreig, 2020-01-27:
Closed by #331

closed 2020-01-27 by adamgreig