stm32-rs/stm32-rs #308: activating Travis CI for stm32-rs repos

pdgilbert, 2019-11-20:
(There seems to be no way to indicate issues for the top level user stm32-rs so I hope someone here can do this.)

Several of the repos under the top level stm32-rs have  .travis.yml files but the Travis CI build has not been activated. (See https://travis-ci.org/stm32-rs.)  In particular, I would like to be able to look at stm32l1xx-hal,  stm32f3xx-hal , and stm32f1xx-hal.

I believe activating these requires someone with admin rights for stm32-rs, so possibly this cannot be done by the people responsible for the individual repos?

adamgreig, 2019-11-20:
Thanks for the heads-up, I've enabled Travis for all the -hal repos. If any of the others also want activating just drop a comment here.

closed 2019-11-20 by adamgreig

jordens, 2019-11-20:
A lot of them are on https://travis-ci.com/stm32-rs (note the .com)