stm32-rs/stm32-rs #326: Issue reading and writing to GPIO on STM32f207

kevswims, 2019-12-31:
# Overview
I am just getting started with evaluating Rust for an embedded application and I am trying to do the "Hello World" of the embedded world of turning on an LED on an ST NUCLEO-F207ZG devboard.

I have run through the following steps and I am seeing some behavior that I cannot explain when trying to use the `stm32f2` crate.

# Issue
My main.rs contains the following code. I am trying to set the PB7 GPIO pin high to turn on the blue LED on the Nucleo board.
```
#![no_std]
#![no_main]

extern crate panic_halt; // you can put a breakpoint on `rust_begin_unwind` to catch panics

use cortex_m_rt::entry;
use stm32f2::stm32f217;
use cortex_m_semihosting::hprintln;

#[entry]
fn main() -> ! {
    let mut peripherals = stm32f217::Peripherals::take().unwrap();

    // The LED is on PB7
    let gpio_b = &peripherals.GPIOB;

    let moder_initial = gpio_b.moder.read().bits();
    let odr_initial = gpio_b.odr.read().bits();

    hprintln!("Moder reset: {:x?}, ODR reset: {:x?}", moder_initial, odr_initial).unwrap();

    gpio_b.odr.modify(|_, w| w.odr7().set_bit());
    gpio_b.moder.modify(|_, w| w.moder7().output());

    let moder_final = gpio_b.moder.read().bits();
    let odr_final = gpio_b.odr.read().bits();

    hprintln!("Moder end: {:x?}, ODR end: {:x?}", moder_final, odr_final).unwrap();

    loop {
        // your code goes here
    }
}
```
## Results
When this runs I get the following printout from the `hprintln`'s and the LED does not come on.
```
Moder reset: 280, ODR reset: 280
Moder end: 280, ODR end: 280
```
## Expected Results
I would expect the final values that I am reading out to be different than the first ones since, as far as I understand, this code should be modifying the register values.

# Detailed steps to reproduce

1. Create project: `cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart`
1. Remove examples: `rm ./examples -r`
1. Modify memory.x
Change flash origin to 0x08000000, length = 1024K
Set flash length = 112K
1. Modify `.cargo/config` 
Set `runner = "arm-none-eabi-gdb -q -x openocd.gdg"`
Set `target = "thumbv7m-none-eabi"`
1. Modify openocd.cfg
Set `source [find interface/stlink.cfg]` Note: This fixes a warning that stlink-v2-1.cfg is deprecated
Set `source [find target/stm32f2x.cfg]`
1. Add `stm32f2` dependency in `Cargo.toml`
    ```
    [dependencies.stm32f2]
    version = "0.9.0"
    features = ["stm32f217", "rt"]
    ```
1. Modify `main.rs` as shown above.
1. Launch `openocd`
1. `cargo run`
1. `continue` in GDB until program fully executes.

adamgreig, 2019-12-31:
Hi, thanks for taking the time to include detailed code and steps to repro, it makes it much easier to see what the issue is.

Here I believe your problem is that you're not enabling the clock to the GPIO peripheral, which means no register access will have any effect and the LEDs won't operate etc.

I'd recommend reading the RCC section of the reference manual for your STM32F207, but the short answer is you probably need something like this:

```rust
let mut peripherals = stm32f217::Peripherals::take().unwrap();
let rcc = &peripherals.RCC;
rcc.ahb1enr.modify(|_, w| w.gpioben().enabled());
```

Hope that helps!

kevswims, 2019-12-31:
Hi @adamgreig, Thanks for the assistance. I was able to get the clocks correctly configured and enable the gpiob as you suggested. With the changes I can successfully control the LED on my board.

It seems to me that the embedded rust book should do more to show a full example of interacting with the hardware in the "hello world" section. I have always considered a blinking LED to be the "hello world" of embedded programming. I will open up an issue over there for suggestions about how I can contribute the example I have created back to the book.

Thanks!

closed 2019-12-31 by kevswims