stm32-rs/stm32-rs #327: Buggy SVD files for STM32F4 USB OTG peripherals

FLamparski, 2020-01-02:
I was trying to do some USB OTG programming. However it looks like not all relevant registers are exposed in this crate, or at least they are mislabelled, owing to some dodgy SVD files. For instance, `FS_DIEPTXF0` appears to be missing from the crate but it looks like it might appear under the name `FS_GNPTXFSIZ`. This might not be the only inconsistency, if I come across any more I'll report them here as well.

FLamparski, 2020-01-02:
`FS_DIEPTXF4` and `FS_DIEPTXF5` appear to be missing completely; only `FS_DIEPTXF{1,2,3}` are exposed and their doc comments are wrong:

```xml
          <name>FS_DIEPTXF1</name>
          <displayName>FS_DIEPTXF1</displayName>
          <description>OTG_FS device IN endpoint transmit FIFO size
          register (OTG_FS_DIEPTXF2)</description>
```

From the manual:

[image]

(RM0390 pp 1143)

FLamparski, 2020-01-02:
`FS_GRXSTSP_Device` - OTG receive status read and pop register - is missing from the SVD. Only `FS_GRXSTSR_Device` is present, and that one does not pop the RX FIFO.

[image]

(RM0390 pp 1132)

Windfisch, 2022-02-07:
There are more bugs. I have fixed those which i found in https://github.com/stm32-rs/stm32-rs/pull/697, and there are probably even more.

Should we close this issue though?

dbristow-otc, 2023-12-31:
I'd like to request that we keep this issue open as a meta-issue for all the bugs in the SVD until we get device/host modes worked out. However, this will be an uphill battle as even the reference manuals seem to be wrong:

[image]
(Page 1291 REM0090 Rev 19) 

Yet ST's official HAL treats this register as writable:

[image]
(STM32CubeF4/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c; line 1526)