stm32-rs/stm32-rs #340: STM32F401 missing TIM4 interrupt

dirk-dms, 2020-02-23:
STM32F401 is missing TIM4 interrupt.
from the reference manual RM0368 page 204:

Position: 30
Priority: 37
Type of priority: settable 
Acronym: TIM4 
Description: TIM4 global interrupt
Address: 0x0000 00B8

dirk-dms, 2020-02-23:
Upon further inspection these interrupt vectors are also missing from STM32F401:
TIM5
DMA1_Stream0
DMA1_Stream1
DMA1_Stream2
DMA1_Stream3
DMA1_Stream4
DMA1_Stream5
DMA1_Stream6
DMA1_Stream7
DMA2_Stream0
DMA2_Stream1
DMA2_Stream2
DMA2_Stream3
DMA2_Stream4
DMA2_Stream5
DMA2_Stream6
DMA2_Stream7

dirk-dms, 2020-02-25:
I'm currently testing the patch modifications for the svd files locally, and will prepare a pull request when tested.

dirk-dms, 2020-02-25:
I made a pull request  #341 with a fix for the missing interrupt handlers waiting for review.

adamgreig, 2020-02-26:
Closed by #341

closed 2020-02-26 by adamgreig