stm32-rs/stm32-rs #344: Why fields are not covered?

MarcoIeni, 2020-02-29:
Hi, I have an stm32f407G-DISC1 and I would like to contribute to the stm32f407 svd.

For example, I was looking at the [RNG](https://stm32.agg.io/rs/STM32F407.html#RNG) peripheral, but I don't understand why the fields of [CR](https://stm32.agg.io/rs/STM32F407.html#RNG:CR) are listed as "not covered" or why only 3 out of 5 field are covered for the [SR](https://stm32.agg.io/rs/STM32F407.html#RNG:SR).
I don't understand what's missing in these fields.

adamgreig, 2020-03-01:
The website called a field "covered" if either it is read-only, or if someone has specified either an enumeration or a range of possible values that can be written.

For example for the STM32H7 we have [some RNG definitions](https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/rng/rng_v1.yaml) to say those bits can either take a 0 or a 1 value and they are called "Disabled" and "Enabled" respectively.

If you wanted to contribute, you could make a similar file for the F407 (which would probably also be useful for many other F4 devices and perhaps others too, any chip that shares a similar RNG peripheral -- the `matchperipherals.py` script can help check for this), where you'd describe the other fields and what values they can possibly take, by looking at the relevant reference manual (RM0090 for the STM32F407 I believe).

MarcoIeni, 2020-03-01:
Thanks, it is clear now :+1:

closed 2020-03-01 by MarcoIeni