stm32-rs/stm32-rs #345: Consider moving what's auto-generated in a different branch/repo

MarcoIeni, 2020-03-01:
In my opinion, it would be clearer to leave in the master branch only the inputs and move the auto generated output in another branch or in another repository.
For what I understood until now, the output consists in the `stm32*` directories.

I have seen that there is the [nightlies](https://github.com/stm32-rs/stm32-rs-nightlies) repository which is in sync with the master.
Maybe there could be a `stable` repository which is updated only when there is a new tag in this repository.

In this way we could remove all the `stm32*` directories from this branch and there is no need to execute scripts like `scripts/makecrates.py` manually.

adamgreig, 2020-03-01:
This is a really good idea, thanks! I don't think there's even any need to make a `stable` repository, since the autogenerated content in this repo is totally useless anyway. We should just make it clearer that the nightlies repo exists and can be used to look at the code.

Would you like to make this change in a PR? If not I can do it this week. I think we'd just need to add the crate directories to `.gitignore` and remove the existing files.

MarcoIeni, 2020-03-02:
> We should just make it clearer that the nightlies repo exists and can be used to look at the code.

I am new to this project and to me it was almost clear.
**Maybe** we could add a sentence to the README at the beginning of [this](https://github.com/stm32-rs/stm32-rs/blob/master/README.md#generating-device-crates--building-locally) section that says something like:

If you just want to look at the crates code, visit the [stm32-rs-nightlies](https://github.com/stm32-rs/stm32-rs-nightlies) repository.
Otherwise, if you have made changes and you want to build the crates locally follow these steps:

adamgreig, 2020-03-04:
Closed by #346

closed 2020-03-04 by adamgreig