stm32-rs/stm32-rs #351: create a `meta` repository

MarcoIeni, 2020-03-10:
You could create a `meta` repository where we discuss about issues related to different repositories of the organization.
I don't know if something like this already exists.
An example is [here](https://github.com/actions-rs/meta).

For example I would like to open an issue about the creation of a standard format/structure for the README files of all the `stm32yyxx-hal` crates, which at the moment have different conventions and structures.
This issue could be opened in this new repository and maybe also the standard itself.

adamgreig, 2020-03-10:
We discussed this a bit in #209 as well but didn't go anywhere. I'm open to the idea but I don't know how much discussion would take place there; I imagined we could have such topics in this repository but none have really come about. Your HAL README example would be a good fit though.

MarcoIeni, 2020-03-11:
Oh, it's a duplicate issue, sorry. However i don't close this because in that issue there are already a lot of topics discussed.
Here we can discuss if a unified structure for the `stm32yyxx-hal` README is needed.
In my opinion it is. I see two alternatives here:
* we discuss about this structure here and once we decide it we just change all the README files. 
* we create and maintain a new readme file that contains the structure. The place for this file should be in a new `meta` repository.

In my opinion the second alternative is better because we have a single source of truth that can be improved with time.

MarcoIeni, 2020-05-01:
The meta repository could also contain the standard ci settings for all the hal crates, see [here](https://github.com/MarcoIeni/stm32f4xx-hal/tree/master/.github/workflows) for example.
Together with PR template, issue template, contributing guidelines and so on.

iostat, 2020-06-17:
+1 for this. I'm interested in creating a set of macros to support generating peripheral initialization code from CubeMX `.ioc` file, and haven't been able to find a sensible place to start a meta-discussion to gauge interest and needs/wants out of it. A meta-repo would be perfect for such a topic

adamgreig, 2020-06-17:
Thanks! I think that's more than enough evidence of demand; I've created [meta](https://github.com/stm32-rs/meta). I'll close this issue now but please feel free to open new ones over there to start working out what should be included in the new repository.

closed 2020-06-17 by adamgreig

burrbull, 2020-06-20:
https://github.com/burrbull/stm32-hals