stm32-rs/stm32-rs #384: STM32L4x2 feature not granular enough

korken89, 2020-05-25:
Hi,

I noticed that the `stm32l4x2` feature pulled in non-existing peripherals for some MCUs in the range.
In my case I noticed it with `TIM3` which only exists for `STM32L451xx`, `STM32L452xx` and `STM32L462xx` while it is also imported for `STM32L412xx`.

Not sure how the architecturally fix this, but I want to highlight the issue.
In my case using the non-existing timer caused a `hardfault`, so it was easy to debug - but maybe other similar issues can be a lot harder to figure out.

adamgreig, 2020-05-25:
I guess the L412 didn't exist at the time; we say that the L4x2 crate [supports L432/42/52/62](https://github.com/stm32-rs/stm32-rs/blob/master/stm32_part_table.yaml#L483-L492) but not 12. Perhaps ST have released a new SVD file for the L412 by itself.

korken89, 2020-05-25:
Right, makes sense.
So maybe it's just to add a new feature that is the "less complex" L4s?

adamgreig, 2020-05-26:
Unfortunately it looks like ST issue the same `STL32L4x2` SVD file for the L412, which evidently isn't quite correct for it.

That said, it's pretty common for devices to be the same except for some missing peripherals, and we haven't generally gone to much effort there. For example, ST release the same SVD for the F405 and F415, which differ only in the CRYP peripheral (as is common for loads of ST parts that come with or without cryptographic hardware). We release the exact same device for both of those, and users on an F405 just have to know to not use the CRYP. I guess it's not ideal but does halve the number of devices we have! :laughing: 

I think for this either we just add the L412 as "supported" under the L4x2 device but users have to not use the peripherals it doesn't have, or you could possibly create a new L412 device which you base on the L4x2 SVD but explicitly remove the non-existent peripherals.

korken89, 2020-05-26:
Haha, well not a big issue IMO - as long as it is known