stm32-rs/stm32-rs #386: Detailed alternate function names for GPIOs

Past9, 2020-06-07:
Is there any interest in having enumerated values for GPIO alt functions that describe the actual function? For example, instead of setting AFRL/AFRH fields to AF1-AF16, values like TIM2_CH1 could be used instead. This info is available in the datasheets, like this one starting on page 45: https://www.st.com/resource/en/datasheet/stm32f303cb.pdf

I'm not sure this isn't already being done, but I don't see any examples of it. Thought I might make a PR for a couple chips I'm using if there's interest.

salkinium, 2020-06-07:
Alternate Function Mapping can be extracted from CubeMX data. See https://github.com/modm-io/modm-devices for a shameless plug ;-P

Past9, 2020-06-07:
Wow, thank you. It looks like the CubeMX data may be incomplete, given that there are some alt funcs missing from those files. For example all but one of the port D alt funcs is missing from the file for the device I linked above: https://github.com/modm-io/modm-devices/blob/develop/devices/stm32/stm32f3-03-6_8.xml

Still, this could save a big chunk of a pretty tedious manual task. Will definitely try to make use of it.

salkinium, 2020-06-07:
The CubeMX data also has some conditions encoded in them, that are resolved with their Java engine. I didn't manage to reverse-engineer that at all, so there's probably more missing. It would be nice to be able to parse the PDFs if only for comparison. I'm flying pretty blind with only manual spot checks here and there.

Past9, 2020-06-07:
I've noticed that some chips in the same family have alt funcs that aren't implemented in others. Like the F303xD/E (https://www.st.com/resource/en/datasheet/stm32f303zd.pdf) have an AF2 for PA9, but the F303xB/C chips don't. So the conditions could be related to that.

Unfortunately I think you'd need OCR software to get anything meaningful from the PDFs. I'm not very familar with that format but I believe the tables aren't actually in a tabular format in the file. It's more like "draw a square here with this text in it, another square there with that text in it, etc." I guess that's what makes it "portable". No ambiguity in how things should be rendered.

adamgreig, 2020-06-07:
The CubeMX data is also available here: https://github.com/stm32-rs/cube-MX-db

I think it would be a _lot_ of work to get a meaningful amount of AF entries encoded compared to what we do right now, but at least in principle it would be nice to encode both what the AF does and also not emit a variant for disconnected AFs.

Right now many of the HALs encode this data instead, so the PAC lets you write any AF, but the HAL has a type-safe constructor for each peripheral which only takes properly connected AFs. for example [here](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/spi.rs#L402-L455). I'm not sure if the HALs could benefit from this information being in the PAC instead: maybe they could have a simpler trait bound if it already wasn't possible to make pins in the "wrong" AF? But they might have to do all the same checking anyway, I'm not sure.

salkinium, 2020-06-07:
> HAL has a type-safe constructor for each peripheral which only takes properly connected AFs

modm has the same, but in C++. `Usart1::connect<GpioA1::Tx, GpioA2::Rx>()` is checked at compile time. You can even check the STM32F1 *remap groups* (one bit remaps multiple pins) with this API.

However the interesting bit of this isn't actually the language implementation, but the data transformation. The whole point of modm-devices was to have *common* data and *common* data transformations (in Python). It's then up to the library developers to "format" this data using the provided transformation into their specific language and HAL flavor.

That would make it possible to transfer the knowledge from one HAL/language to the other without having to port everything again.

> the tables aren't actually in a tabular format in the file

That's true, the tables are drawn via a Postscript like language and the text is placed character by character. You don't need OCR however, since it's vector graphics, you "just" needs to interpret the rectangles as table lines. The pdfium library provides a C++ API for this, I've written a (very simple) [X-Ray PDF viewer](https://github.com/salkinium/qtdde) some years ago, just got side-tracked with work/uni. It's probably not much work to translate the PDF first to a flattened HTML, on which you can then do actual semantic transformations (instead of dealing with 2D graphics).

It would be very useful to extend/correct the SVD with the RM/datasheet data instead of doing it manually like now.

[image]
[image]

salkinium, 2020-06-07:
Oh, the thing it still compiles… Here is the AF table with graphics in blue and char bounding box in red.
[image]

Past9, 2020-06-09:
I made a PR for the F303's to show what I had in mind:
https://github.com/stm32-rs/stm32-rs/pull/388

 I left all the disconnected alt funcs in place with their generic names for now.

I just filled these in by hand from the datasheet, which I'd estimate was 2-3 hours of work. Like @adamgreig said, this would be a lot of work to do for all devices. Not sure if it would be acceptable to let contributors do each device as needed/desired, or if they would all need to be done in one effort.

I'd love to parse the PDFs for this information but unfortunately I lack the time and knowledge to implement that myself.

closed 2025-05-27 by burrbull

FeldrinH, 2025-07-20:
> I think it is better to make it in HALs. Something like https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/gpio/alt/f4.rs is much more useful then enums in PACs. Reopen if you do not agree.

Automating the generation of such mappings separately for each HAL seems like a lot of duplicate work. I'm not sure how exactly the mappings should be exposed, but (for now) the PAC is much better positioned to automate this for all STM32 families together.