stm32-rs/stm32-rs #387: Try to load the into blue pill project

yehielliv, 2020-06-07:
Trying to build with blue pill project I get:
> Executing task in folder stm32-blue-pill-rust: cargo build --release <

    Updating git repository `https://github.com/stm32-rs/stm32-rs`
error: failed to load source for a dependency on `stm32f103xx`

Caused by:
  Unable to update https://github.com/stm32-rs/stm32-rs

Caused by:
  Could not find Cargo.toml in `C:\Users\KNC\.cargo\git\checkouts\stm32-rs-19318749ec3518a9\1b63ec8`
The terminal process terminated with exit code: 1

My cargo.tomlis attached.
Any idea how to solve ths error.

Thank you.

[Cargo.zip](https://github.com/stm32-rs/stm32-rs/files/4743215/Cargo.zip)

adamgreig, 2020-06-07:
Hello,

It looks like you are using a very old template which is out of date, but have changed the old stm32f103xx and stm32f103xx-hal to point to new stm32-rs repositories which do not contain those crates at all.

You probably want something more like:

```toml
[dependencies]
cortex-m = "0.6"
cortex-m-rt = "0.6"
stm32f1xx-hal = { version = "0.6", features = ["rt", "stm32f103", "medium"] }
cortex-m-semihosting = "0.3"
panic-semihosting = "0.5"
```

Check out some more examples and documentation on the HAL: https://github.com/stm32-rs/stm32f1xx-hal/

adamgreig, 2020-06-07:
Hi, you might also try this out: https://github.com/reneherrero/blue-pill-quickstart

yehielliv, 2020-06-07:
Thanks.
That did not work so I tried 
[dependencies]
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.2"
cortex-m-rt = "0.6.11"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"

[dependencies.stm32f1xx-hal]
version = "0.6.0"
features = ["rt", "stm32f103", "medium"]
which got this error
Downloading stm32f1xx-hal v0.6.0
error: unable to get packages from source

Caused by:
  failed to parse manifest at `C:\Users\KNC\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f1xx-hal-0.6.0\Cargo.toml`

yehielliv, 2020-06-07:
Trying the 2nd suggetion again got

    Updating git repository `https://github.com/stm32-rs/stm32-rs`
error: failed to load source for a dependency on `stm32f103xx`

Caused by:
  Unable to update https://github.com/stm32-rs/stm32-rs

Caused by:
  Could not find Cargo.toml in `C:\Users\KNC\.cargo\git\checkouts\stm32-rs-19318749ec3518a9\1b63ec8`
The terminal process terminated with exit code: 1

adamgreig, 2020-06-07:
What version of Rust are you using? (`rustc --version`).

You shouldn't have "https://github.com/stm32-rs/stm32-rs" anywhere in Cargo.toml, because it does not contain any crates. None of those examples have it either though, so maybe it's been left over? Could you paste your entire Cargo.toml?

yehielliv, 2020-06-08:
Thank you for your help.
I'm using vs codewith rust 0.7.8 as an extension.
The blue pill project cargo.toml code is
# This file declares the dependencies for the Rust program.
[package]
name = "stm32-blue-pill-rust"
version = "0.3.2"
authors = ["Lee Lup Yuen <luppy@appkaki.com>"]
description = "A sample Rust application for STM32 Blue Pill microcontrollers"
keywords = ["arm", "cortex-m", "stm32", "bluepill"]
categories = ["embedded", "no-std"]
license = "MIT OR Apache-2.0"
repository = "https://github.com/lupyuen/stm32-blue-pill-rust"

# To minimize the executable size, we should always compile with --release flag. 
[profile.release]
# To optimize the usage of many crates, we should link with Link Time Optimization (LTO). 
lto = true
codegen-units = 1
# Enable debugging in release mode.
debug = true

[dependencies]
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.2"
cortex-m-rt = "0.6.11"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"

[dependencies.stm32f1xx-hal]
version = "0.6.0"
features = ["rt", "stm32f103", "medium"]

and the quick start cargo.toml is
[package]
authors = ["Your Name"]
edition = "2018"
readme = "README.md"
name = "blue-pill-quickstart"
version = "0.1.0"

[dependencies]
stm32f1xx-hal = { version = "0.5.3", features = ["rt", "stm32f103" ] }
cortex-m = "0.6.2"
cortex-m-rt = { version = "0.6.12", features = ["device"] }
panic-semihosting = "0.5.3"

# this lets you use `cargo fix`!
[[bin]]
name = "blue-pill-quickstart"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations

burrbull, 2020-06-08:
update rust with `rustup update`.
```
cargo clean
cargo update
```
and show what shows
`cargo tree`

and use triple backquoters when place code here

yehielliv, 2020-06-09:
Even after the suggetion steps that did update rust and cargo, still I get
 Executing task in folder stm32-blue-pill-rust: cargo build --release <

error: unable to get packages from source

Caused by:
  failed to parse manifest at `C:\Users\KNC\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-0.6.2\Cargo.toml`

Caused by:
  editions are unstable

I checked anf the cargo.toml does exist in that directory.
Any idea why it does not find it or parse it?

Thank you.

adamgreig, 2020-06-10:
> editions are unstable

It sounds like you might still be running a very old version of Rust. Could you run `rustup update` and then paste the output of `rustc --version`?

closed 2020-06-10 by yehielliv