stm32-rs/stm32-rs #397: Recommended way of depending on unreleased features rapidly breaks

cbiffle, 2020-07-06:
We're using some of the unreleased stm32h7 support in this crate. Based on README, we specified a dependency on the `stm32-rs-nightlies` repo, since the non-nightly repo seems to requires some additional steps/materials to build.

It looks like, because the commits in nightlies aren't tagged or otherwise retained, they're getting gc'd fairly rapidly. The commit we picked up in our `Cargo.lock` last week is already gone, breaking our builds.

I see no tags or other stable heads in the repo. Which commits did you intend for people to depend on?

adamgreig, 2020-07-06:
The "nightly" builder pushes a new commit on each merge to master without storing any historic commits, mainly to prevent the repo size blowing up as the generated files (and diffs) tend to be large and have to be downloaded by everyone cloning them. We could possibly move to an approach where the last n commits are kept instead, but it would only postpone the problem. Potentially we could move to keeping all commits since the latest release or some similar compromise, though the implementation complexity would be a bit higher. There don't tend to be _that_ many merges to master between releases, maybe 10-30, but the nature of the generated files is such that it can still represent a significant total filesize.

My hope was that users could always use the latest (and only) commit in the master branch, but I don't know if it's possible for a lock file to specify anything other than a unique commit hash, in which case end user CI builds with lock files (such as yours) suffer when that hash is GC'd.

For what it's worth, building the crate from the sources (i.e. this repository) is not especially difficult, just slow and can't be done entirely by Cargo. The CI runner for it is [here](https://github.com/stm32-rs/stm32-rs/blob/master/.github/workflows/nightlies.yaml).

cbiffle, 2020-07-06:
> My hope was that users could always use the latest (and only) commit in the master branch

That opens things up to a _different_ kind of build failure, where something changes in the upstream repo that breaks us in a subtle way in an unrelated build. We pin everything from the toolchain down.

So it sounds like there is no robust way to depend on the nightlies repo at this time, short of maintaining a fork with tags. I can do that.

> For what it's worth, building the crate from the sources (i.e. this repository) is not especially difficult, just slow and can't be done entirely by Cargo.

That's a little impractical for projects with stm32-rs as a Cargo dependency; we'd need to e.g. check it out as a submodule, patch the registry to cover transitive dependencies, and have every dev follow those instructions for every new or updated clone of the source.

adamgreig, 2020-07-06:
> So it sounds like there is no robust way to depend on the nightlies repo at this time, short of maintaining a fork with tags

What specifically do the the tags do, just ensure a given commit isn't GC'd?

> That's a little impractical for projects with stm32-rs as a Cargo dependency; we'd need to e.g. check it out as a submodule, patch the registry to cover transitive dependencies, and have every dev follow those instructions for every new or updated clone of the source.

Oh, for sure. I was more imagining you could set up a repo which kept your own built versions hosted relatively easily. Basically our nightlies repo except updated when you update your dependency version instead of whenever there's a push to master, I guess.

cbiffle, 2020-07-06:
> What specifically do the the tags do, just ensure a given commit isn't GC'd?

Yes.

> I was more imagining you could set up a repo which kept your own built versions hosted relatively easily. Basically our nightlies repo except updated when you update your dependency version instead of whenever there's a push to master, I guess.

https://github.com/oxidecomputer/stm32-rs-nightlies :-)

adamgreig, 2020-07-09:
I've added a note to 39dee00 suggesting users fork the nightlies repo if they need a stable commit hash. I think that's a nice solution, since it keeps the commit they're using locked until they manually update their fork, without impacting any other nightly users, and it's very easy to do in GitHub. Hopefully it's working out for you too.

closed 2020-07-09 by adamgreig