stm32-rs/stm32-rs #399: JEXTSEL incorrect for some (all?) adc_v2 devices

stabler, 2020-07-11:
The current JEXTSEL mapping for adc_v2 has ([link](https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/adc/adc_v2.yaml#L92)):
```
    JEXTSEL:
      TIM1TRGO: [0, "Timer 1 TRGO event"]
      TIM1CC4: [1, "Timer 1 CC4 event"]
      TIM2TRGO: [2, "Timer 2 TRGO event"]
      TIM2CC1: [3, "Timer 2 CC1 event"]
      TIM3CC4: [4, "Timer 3 CC4 event"]
      TIM4TRGO: [5, "Timer 4 TRGO event"]
      TIM8CC4: [7, "Timer 8 CC4 event"]
      TIM1TRGO2: [8, "Timer 1 TRGO(2) event"]
      TIM8TRGO: [9, "Timer 8 TRGO event"]
      TIM8TRGO2: [10, "Timer 8 TRGO(2) event"]
      TIM3CC3: [11, "Timer 3 CC3 event"]
      TIM5TRGO: [12, "Timer 5 TRGO event"]
      TIM3CC1: [13, "Timer 3 CC1 event"]
      TIM6TRGO: [14, "Timer 6 TRGO event"]
```

The reference manual for STM32F40x -> 43x ([RM0090](https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf), section 13.13.3) and 44x ([RM0390](https://www.st.com/resource/en/reference_manual/dm00135183-stm32f446xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), section 13.13.3) has the ordering as:
```
    JEXTSEL:
      TIM1CC4: [0, "Timer 1 CC4 event"]
      TIM1TRGO: [1, "Timer 1 TRGO event"]
      TIM2CC1: [2, "Timer 2 CC1 event"]
      TIM2TRGO: [3, "Timer 2 TRGO event"]
      TIM3CC2: [4, "Timer 3 CC2 event"]
      TIM3CC4: [5, "Timer 3 CC4 event"]
      TIM4CC1: [6, "Timer 4 CC1 event"]
      TIM4CC2: [7, "Timer 4 CC2 event"]
      TIM4CC3: [8, "Timer 4 CC3 event"]
      TIM4TRGO: [9, "Timer 4 TRGO event"]
      TIM5CC4: [10, "Timer 5 CC4 event"]
      TIM5TRGO: [11, "Timer 5 TRGO event"]
      TIM8CC2: [12, "Timer 8 CC2 event"]
      TIM8CC3: [13, "Timer 8 CC3 event"]
      TIM8CC4: [14, "Timer 8 CC4 event"]
      EXTILINE15: [15, "EXTI line15"]
```

I suspect this error also exists in adc_v3 ([link](https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/adc/adc_v3.yaml#L158)) but I haven't confirmed against a data sheet.

I'm not sure if this is a copy-paste error from a previous generation device, or if there's inconsistency within the device family with other parts that I didn't check.

adamgreig, 2020-07-12:
Thanks for catching this! I think most likely this is being applied too widely to devices with different ordering. If you'd like to try and fix it I'd welcome a PR correcting the order for the appropriate devices, otherwise I'll try and get it sorted before the next release.

FeldrinH, 2026-01-22:
Is this now fixed? The JEXTSEL values for adc_v2 have been split into a number of different files and at least one of them ([adc_v2_extsel_c.yaml](https://github.com/stm32-rs/stm32-rs/blob/master/devices/fields/adc/adc_v2/adc_v2_extsel_c.yaml)) does correspond to the RM0090 and RM0390 values quoted here.